prusa3d/PrusaSlicer #3669: Suboptimal slicing of thin-wall objects

normanhendrich, 2020-02-16:
Hi there,

the default slicing of objects with thin walls is suboptimal, simple test object attached.
For structures at least 2 extrusions widths wide the silcing seems to work, but thinner structures are sliced in a somehwat "annoying" way, almost guaranteeing that the final object will break. Maybe I just missed the correct slicing settings? 

Thanks for any advice,
  Norman

### Version
PrusaSlicer-2.2.0-alpha2-linux-x64-2019122221342 basen on Slic3r

### Operating system type + version
Linux Ubuntu 16.04

### 3D printer brand / version + firmware version (if known)
This is printer independent, the printer works. Just the slicing is suboptimal.

### Behavior
* I would like to slice an object with a thin 1-perimeter wall, to be used as a very low-force spring.
* The spring is mounted between more solid base objects.

* Simplified test object attached (scad, stl, 3mf). The preferred solution would be a continuous extrusion along the lenght of the thin part, with start and stop somewhere inside the thick cubes.

* Whatever settings I try, the slicing always creates nice solid perimeters around the base objects,
  and then attaches the spring in separate extrusions. Printed this way, the object cannot even be
  removed from the printbed without breaking.

* _Expected Results_
* _Actual Results_

Suboptimal slicing, with single extrusion path broken into several pieces. Object much more fragile than possible. 

[prusa_slicer_hates_one_perimeter_objects.zip](https://github.com/prusa3d/PrusaSlicer/files/4209566/prusa_slicer_hates_one_perimeter_objects.zip)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

* Is this a new feature request? Probably yes. I would LIKE to have 1-perimeter extrusions running
 as long as possible, preferring continuous extraction in THIN parts and any extrusion stops and restarts anywhere in THICK objects where it doesn't really matter.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while 
all other files need to be zipped first (.zip, .gz)_

cubiq, 2020-02-18:
I noticed that too and it is indeed annoying. You can "fix" that by reducing the outside perimeter width by a small amount (usually -0.02 is enough) but you end up with a little under extrusion.

rtyr, 2020-04-18:
You can fix this particular model by disabling "detect thin walls" feature. Other that that, it is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/149, https://github.com/prusa3d/PrusaSlicer/issues/227, etc. Closing this one.

closed 2020-04-18 by rtyr