prusa3d/PrusaSlicer #3670: "Reload from disk" loads updated part BELOW the build plate (v2.1.1+)

mkoistinen, 2020-02-16:
### Version
PrusaSlicer-2.1.1+-201912101508

### Operating system type + version
MacOS Catalina 10.15.3

### 3D printer brand / version + firmware version (if known)
Original Prusa mk3s + mmu2

### Behavior
* When using "Reload from disk" to update a part, the z-position is not adjusted to accommodate new part heights. This can result in some of the part being "below" the build plate and not printing correctly.
* Create a part, export as STL
  * Load STL into PrusaSlicer
  * Increate the height of the part in the modeling software and re-export as STL
  * Use "Reload from disk" in PrusaSlicer
  * Note that the part is no longer resting on the build place
* The z-position should be adjusted so that the part is resting on the build plate
* In my case, the part "Reload(ed) from disk" and had 3 mm of my part below the build plate

[image]

mkoistinen, 2020-02-16:
Note, I see that "Reload from disk" has been reworked in 2.2.0 alphas, but I don't see anything in the release notes about recalculating the z-position of the part. I hope someone can ensure that the new version of "Reload from disk" correctly adjusts the z-positions of the parts too.

FidelCapo, 2020-02-18:
Thank you for report. This issue should be fixed in next release.
Closing

closed 2020-02-18 by FidelCapo