prusa3d/PrusaSlicer #3671: Perimeters are being made where they shouldn't

GWdd, 2020-02-18:
### Version
2.1

### Operating system type + version
Win 10 1909

### 3D printer brand / version + firmware version (if known)
i3-MK3

### Behavior
* _Describe the problem_
In an attempt to create a part that is a perimeter shell with no top or bottom layers, Prusa Slicer interprets a curved recess as a wall rather than a top surface, and adds perimeters.

Adding a modifier with no perimeters helps, but leaves artifacts - a single line extrusion in some places, and almost a full single layer extrusion in other places, all where PrusaSlicer really wants to build a full perimeter.

[image]

* _Steps needed to reproduce the problem_

* _Expected Results_
A cylinder with hex infill, not top or bottom layers. A spherical 'cutout' or 'void' at the top.

* _Actual Results_
Spherical surface has perimeters; adding modifiers doesn't seem to remove them.

 * _Screenshots from __*PrusaSlicer*__ preview are preferred_
 See attached.

_Is this a new feature request?_
Nope.

#### Project File (.3MF) where problem occurs
[Adapter.zip](https://github.com/prusa3d/PrusaSlicer/files/4218240/Adapter.zip)

bubnikv, 2020-02-20:
What you are seeing are the usual infill connecting lines following the subtracted volume. Why do you think this is an incorrect behavior?

Pun-e, 2020-02-20:
Yeah, those aren't perimeters, they're connecting elements between the infill elements.

I understand how you could be surprised by them, but that's standard behaviour.

[image]

You can see them in this random part I'm printing. I would also appreciate being able to turn them off when I want to make an infill-only part, but not a big enough issue for me to file a ticket over

GWdd, 2020-02-20:
The desired outcome is a honeycomb infill, with nothing on top (i.e., no top layers what so ever).

GWdd, 2020-02-20:
What's making me think it is a defect is the fact some of the hex patterns get a line or two, some get 25%, 50% and even 100% covered.  This doesn't sound like an infill characteristic.

GWdd, 2020-02-20:
Perhaps a better question is: if I print a sphere, when does a perimeter become a top surface?

GWdd, 2020-02-20:
Thinking more about it: they are a defect because there is no perimeter to connect to: essentially the slicer is making lines where none should be. No perimeter, no infill connections is the 'proper' behavior.

joshzume, 2020-04-23:
"connect infill lines" as an option would be very helpful in certain multi-body models.

bubnikv, 2020-11-30:
In PrusaSlicer 2.3.0-beta1 you can disable the infill anchors
[image]

[image]

That should fulfill your needs.

closed 2020-11-30 by bubnikv