prusa3d/PrusaSlicer #3674: Preset for Prusament PETG is lower than range - causes issues

stoofer13, 2020-02-19:
### Version
2.1.1

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3S/MMU2S

### Behavior
Prusament PETG has 250C +/- 10C printed on it for an effective range of 240C-260C.

Prusament PETG (also non-Prusament) MMU2 preset in Slicer is 230C.

In the wipe tower when using PVA support @ 205C, I am experiencing the PETG having very poor adhesion at preset value, resulting in strings of PETG, balling, and wipe tower failing.  At 250C, for the Prusament PETG, the print completes fine (with a little stringing) and the wipe tower is in no danger of failing.

It seems the preset for the Prusament PETG is too cold at 10C lower than the advertised range in this scenario.

#### Project File (.3MF) where problem occurs
NA

wpegden, 2020-02-19:
Not sure if this is a bug per se.  Certainly, the lower temperature is intentional because of presumed tradeoffs with loading reliability.

There are some other settings you can play with to get better purge towers.  See for example andre-l9's post here:
https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-assembly-and-first-prints-troubleshooting/purge-tower-tuning/

stoofer13, 2020-02-19:
Technically it's a "Failure" caused by a "Fault" in PrusaSlicer or the configurations of the System Presets. Just because the value was deliberately entered at 230 "by design" doesn't make it correct :)

I believe the fix is to change the System Preset to be hotter and/or change default purge tower settings for better adhesion.  Where Prusa is in control of the whole chain (Printer, software, configurations and filament, this is not an assembly issue) it should work - any other result is a failure caused by a fault.

wpegden, 2020-02-19:
Personally I use 240 or 245 when printing PETG with the MMU2S.  That together with those purge tower settings I mentioned helped a lot.

rtyr, 2020-02-19:
Yes, it is by design to improve reliability. You can of course use higher temperature if it works better for you.

stoofer13, 2020-02-19:
Just because it is "by design", does not mean that it is not a fault. If I were not using Prusament, I would not be reporting this, but everything I am using is from the Prusa ecosystem. It should work without me changing any settings - any other result indicates a fault.

bubnikv, 2020-03-15:
as commented by @rtyr, this has been done intentionally and you can change the temperature yourself.
Closing.

closed 2020-03-15 by bubnikv