prusa3d/PrusaSlicer #3675: Bug: Changing print settings resets filament settings

Pun-e, 2020-02-19:
### Version
[image]

### Operating system type + version
W10 Pro 64

### Behavior
Change any option in filament settings. E.g. First layer bed temp:
[image]

Then change print settings to another option:
[image]

And now our previous change is gone *without any warning*
[image]

There's nothing wrong with PS having this behaviour, but if this is expected then the user needs to be notified. Otherwise it might just be a bug that's flown under the radar. Very annoying bug to discover for yourself.

Edit:
Did some testing, and this *only* affects filament settings, printer settings aren't reset the same way

bubnikv, 2020-02-19:
It seems to work fine in 2.2.0-alpha4. Would you please retest with
2.2.0-alpha4? Thanks.

Pun-e, 2020-02-19:
Just installed 2.2A4 and can confirm that I still have the same issue. 

Any configuration of custom/default print & filament settings causes this issue so I don't think it's anything to do with any of my custom profiles

bubnikv, 2020-02-19:
Maybe you can share your configuration folder with us, so we can test on your data?

wpegden, 2020-02-19:
This happens for me too, if I run alpha2 out of the box from an appimage on Ubuntu with no config changes.

Obviously if I save the filament settings it's not a problem.

But if I leave it as Prusament PLA (modified) and then change the print setting, then the Prusament PLA (modified) is lost and it uses the unmodified Prusament PLA profile.

bubnikv, 2020-02-19:
@wpegden does it happen every time? Or just after applicaton start? Or when the moon is high?

wpegden, 2020-02-19:
It seems to happen every time.  Moon is just setting here.

bubnikv, 2020-02-20:
> But if I leave it as Prusament PLA (modified) and then change the print setting, then the 
> Prusament PLA (modified) is lost and it uses the unmodified Prusament PLA profile.

I see. I can reproduce this scenario and it is certainly wrong. Thanks for heads up. We should fix that.

Pun-e, 2020-02-20:
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/4228547/PrusaSlicer_config_bundle.zip)

Are these the files you're looking for?

bubnikv, 2020-02-21:
Fixed with 004b23e362e38154197401148224565f1ad5f305

    When switching a Print profile, the modifications of an active Filament
    profile were incorrecly dropped even if the active Filament profile
    was compatible with the newly selected Print profile.

Thanks for heads up. The bug was there for a very long time I suppose.
Closing. The fix will be part of PrusaSlicer 2.2.0-beta.

closed 2020-02-21 by bubnikv