prusa3d/PrusaSlicer #3676: [2.2-aplha] disrupting Serial Port

FlailAway, 2020-02-19:
### Versions
2.1.0+linux64
2.2.0-alpha4+linux-x64

### Operating system type + version
Mint 19.3 + Cinnamon

### 3D printer brand / version + firmware version (if known)
MKS-Gen-L
Marlin 1.1.9

### Behavior
Disrupts Serial connection for other running programs.

I do not know where this is happening within PS. I have a program running that is constantly recieving data from an Arduino Nano connected via the Serial port. If that Serial port drops out for some reason then my program pops up a MessageBox and shows it has lost the connection.

Shortly after I start PS this drop-out happens so I have to leave the program off line until I have finished with the slicing and then restart my program. My program will do 2 recovery Serial connection attempts before declaring it lost. It seems PS is constantly checking the Serial port as it will swamp those 2-recovery attempts.

In the "Printer" setup could there be a "None" in the "Octopi etc" drop down that would disable all attempts at PS checking the Serial ports? Or some other form of preventing this?

Thanks

bubnikv, 2020-02-20:
Are you sure PrusaSlicer is accessing any serial ports?

PrusaSlicer enumerates HID (Human Interface Devices) every 2 seconds in
bool Mouse3DController::connect_device()
calling
hid_enumerate(0, 0);

to detect for the 3D mouse. This may or may not be the culprit of your
issues.

Another hypothesis of mine is that your computer gets loaded (the bus may
get blocked, bad drivers may block) by the graphics processing or something
else.

We may try to create a build of PrusaSlicer, where we would suppress the
HID enumeration to find out whether it helps in your case.

kocikdav, 2020-02-20:
@FlailAway is this issue for both versions you have mentioned? If this distruption happens for version 2.1.0, that would eliminate possibility of hid_enumerate() being the problem, since that was added at 2.2.0-alpha1.

FlailAway, 2020-02-20:
> Are you sure PrusaSlicer is accessing any serial ports?

Thanks for the follow up. The answer to that is yes and no, it is an assumption, but one based on repeatability. I am using workspaces (WS) in Mint 19. I always load my Serial program-Arduino in WS-#4, in fact it is loaded and runs automatically on start up and that PC is generally always running 24/7/365.

I go to another WS, let's say WS-#2 and load PrusaSlicer, use it to slice and save a GCode file and then close PrusaSlicer. There is always the MessageBox from my program in WS-#2 (or whichever WS is current other than WS-#4) showing that the Serial port has been dropped. 

The fact that my program is in WS-#4 and the MessageBox is always in the active WS that I have just used PrusaSlicer in suggests to me that the Serial port is being messed with by it.

I cannot categorically say it is PrusaSlicer, but it happens every time I use it. It does seem to happen an indeterminate time after launching PrusaSlicer. It does not seem to happen instantly I launch PrusaSlicer, but a little time after. I have no idea when as the MessageBox does not rise to the top.

Yes, it does it in 2.1 as I have just begun using PrusaSlicer and started with 2.2 but thought I'd drop back to the stable 2.1 after the first few times it happened with 2.2.

bubnikv, 2020-02-20:
*If it happens with PrusaSlicer 2.1.1 as well, then I don't think
PrusaSlicer is guilty. PrusaSlicer 2.1.1 only accesses serial ports from
the firmware update dialog.*

FlailAway, 2020-02-20:
Ummm, it only happens when using PrusaSlicer. I have been running this program+Arduino for 3+years. If it is any consolation, Cura 4.4.1 and Repetier all do the same thing. 

There is definitely some messing with the Serial ports going on. I have the same program+Arduino on this PC and it has not thrown the error to maybe a year or more. 

Both are running the same Mint, both have i5, both have 8GB memory and both have SSDs. The only difference is that I do not do 3D printing on this PC, only the other one.

And three days back I deleted all instances of Cura and Repetier, did a Timeshift Restore to 3-weeks back and clean Mint install before using the Prusa appimage. Now it is doing the same thing. I have not used Prus this morning and there has been no error.

I know it is all circumstantial, but it is pretty convincing as I see it. I will do no slicing on the other PC for a few days more and I am certain there will be no error message.

Are there any logs I can check or track with?

bubnikv, 2020-02-21:
> Ummm, it only happens when using PrusaSlicer. I have been running this
program+Arduino for 3+years. If it is any consolation, Cura 4.4.1 and
Repetier all do the same thing.

Are you saying that you have the same issue with the serial ports if you
are using Cura or Repetier as if you are using PrusaSlicer? Or are you
saying that PrusaSlicer blocks access of Cura or Repetier to serial ports?

FlailAway, 2020-02-21:
@bubnikv 
Yes, both Cura and Reptier bully the serial ports. Nothing else I use does this. If I do no 3D printing stuff for weeks then the program+Arduino runs fine. I also use that PC on a CNC Lathe, a 50W Laser engraver and CNC milling machines and have no issue as none of them require Serial port access. I am in no doubt that it is s0mething to do with applications that require the Serial port. Maybe it is something somewhere in the setup preamble of stuff that uses Serial.

I am trying to isolate Prusa and/or Cura from the rest of the PC, but VMware and VirtualBox both need an active Internet connection to call home and that PC does not have one. I bring it into the house when I need to update Mint etc, but for normal work in the garage it has no connection.

I have tried using Gnome-Boxes but it is too old (2012) and I have not yet found a Linux OS that will run in a Box. I tried Ubuntu 12.xx but no luck. Mostly a video driver conflict with the 2-year old PC and 2012 Gnome-Boxes.

I really want to make this work as PrusaSlicer is by far the best slicer I have ever used and that includes ALL of the slicers I paid a bunch of money for. Keep up the great work, but I have to get around this Serial trouble somehow.

bubnikv, 2020-02-21:
From what you have described it looks to me that your 3D graphics driver is
faulty as it blocks the other drivers, therefore you are losing data on
your serial ports.
I am afraid we cannot help here.

I have a similar experience with a faulty intel WiFi driver on one of my
older Windows box, which blocks real time audio, so that the real time
audio does stutter.

bubnikv, 2020-03-15:
I am afraid we cannot help here. Closing.

closed 2020-03-15 by bubnikv