prusa3d/PrusaSlicer #3677: Crash when slicing a large item

triantos, 2020-02-20:
### Version
2.1.1+

### Operating system type + version
Macos 10.15.3 (Catalina)

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2. MK3S firmware: 3.8.1-2869, MMU2 firmware 1.0.6-372

### Behavior
* App crash when slicing a particular STL
* Repro steps:
  * Download thing from https://www.thingiverse.com/thing:37117
  * Load Koch_SnowflakeTwistySmoothTribase.stl from that download package (ZIP of STL attached)
  * Once placed in PrusaSlicer, scale to 170% (in all 3 directions)
  * Note the object size is 215mm x 197mm x 194mm, which should fit
  * Click the "Slice now" button
* Crash log attached in the text file

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[vase.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4228469/vase.3mf.zip)
[prusa-crash.txt](https://github.com/prusa3d/PrusaSlicer/files/4228471/prusa-crash.txt)
[Koch_SnowflakeTwistySmoothTribase.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4228474/Koch_SnowflakeTwistySmoothTribase.stl.zip)

triantos, 2020-02-20:
The crash does not happen if I don't scale up the model

bubnikv, 2020-02-20:
Slicer runs out of memory on your computer with this particular model, see the attached memory log.

[memory-log.txt](https://github.com/prusa3d/PrusaSlicer/files/4229834/memory-log.txt)

Namely

```
[2020-02-20 10:02:21.675713] [0x00001e24] [info]    Exporting G-code... WorkingSet: 351MB; PrivateBytes: 301MB; Pagefile(peak): 301MB(311MB)
[2020-02-20 10:02:21.685704] [0x00003640] [debug]   Loading print object toolpaths in parallel - start (GLVolumeCollection RAM: 2MB GPU: 0MB Both: 0MB) WorkingSet: 351MB; PrivateBytes: 303MB; Pagefile(peak): 303MB(311MB)
[2020-02-20 10:02:23.135830] [0x00003640] [debug]   Loading print object toolpaths in parallel - finalizing results (GLVolumeCollection RAM: 1,908MB GPU: 0MB Both: 0MB) WorkingSet: 2,259MB; PrivateBytes: 2,216MB; Pagefile(peak): 2,216MB(2,258MB)
[2020-02-20 10:02:26.032101] [0x00003640] [debug]   Loading print object toolpaths in parallel - end (GLVolumeCollection RAM: 1,908MB GPU: 1,905MB Both: 1,905MB) WorkingSet: 4,164MB; PrivateBytes: 4,123MB; Pagefile(peak): 4,123MB(4,123MB)
[2020-02-20 10:02:37.981852] [0x00001e24] [trace]   Exported layer 0 print_z 0.2, time estimator memory: 0MB, analyzer memory: 0MB WorkingSet: 4,179MB; PrivateBytes: 4,139MB; Pagefile(peak): 4,139MB(4,139MB)
```

and then at the end 
```
[2020-02-20 10:13:12.582395] [0x00001e24] [trace]   Exported layer 1289 print_z 193.55, time estimator memory: 1,913MB, analyzer memory: 1,203MB WorkingSet: 5,823MB; PrivateBytes: 7,358MB; Pagefile(peak): 7,358MB(8,522MB)
[2020-02-20 10:13:13.399903] [0x00001e24] [trace]   Exported layer 1290 print_z 193.7, time estimator memory: 1,913MB, analyzer memory: 1,203MB WorkingSet: 5,824MB; PrivateBytes: 7,358MB; Pagefile(peak): 7,358MB(8,522MB)
[2020-02-20 10:13:14.203438] [0x00001e24] [trace]   Exported layer 1291 print_z 193.85, time estimator memory: 1,913MB, analyzer memory: 1,203MB WorkingSet: 5,824MB; PrivateBytes: 7,358MB; Pagefile(peak): 7,358MB(8,522MB)
[2020-02-20 10:13:15.833478] [0x00001e24] [debug]   Time estimator post processing WorkingSet: 5,825MB; PrivateBytes: 7,358MB; Pagefile(peak): 7,358MB(8,522MB)
[2020-02-20 10:13:39.125231] [0x00001e24] [debug]   Preparing G-code preview data WorkingSet: 5,825MB; PrivateBytes: 7,358MB; Pagefile(peak): 7,358MB(8,522MB)
[2020-02-20 10:13:41.321886] [0x00001e24] [info]    Exporting G-code finished WorkingSet: 4,735MB; PrivateBytes: 6,177MB; Pagefile(peak): 6,177MB(8,522MB)
[2020-02-20 10:13:42.232336] [0x00003640] [debug]   Loading G-code extrusion paths - start (GLVolumeCollection RAM: 0MB GPU: 0MB Both: 0MB) WorkingSet: 330MB; PrivateBytes: 443MB; Pagefile(peak): 443MB(8,522MB)
[2020-02-20 10:13:42.240332] [0x00003640] [debug]   Loading G-code extrusion paths - create volumes (GLVolumeCollection RAM: 0MB GPU: 0MB Both: 0MB) WorkingSet: 330MB; PrivateBytes: 443MB; Pagefile(peak): 443MB(8,522MB)
[2020-02-20 10:13:42.252327] [0x00003640] [debug]   Loading G-code extrusion paths - populate volumes (GLVolumeCollection RAM: 16MB GPU: 0MB Both: 0MB) WorkingSet: 330MB; PrivateBytes: 457MB; Pagefile(peak): 457MB(8,522MB)
[2020-02-20 10:13:47.042440] [0x00003640] [debug]   Loading G-code extrusion paths - end (GLVolumeCollection RAM: 1,891MB GPU: 1,859MB Both: 1,859MB) WorkingSet: 4,064MB; PrivateBytes: 4,194MB; Pagefile(peak): 4,194MB(8,522MB)
```

bubnikv, 2020-02-20:
The slicing took 301 MB, 
the initial G-code preview took 3838 MB, this is mostly the GPU data.
time estimator data and the G-code analyser data took 3219 MB together, raising the total memory allocated to 7358 MB. This is the peak.
Then the initial G-code preview (GPU data) got released and replaced with the final G-code preview, and the G-code analyser data got released, dropping down to 4194 MB. This is the final RAM consumption on my system.

We can likely reduce memory consumption in the time estimator.

enricoturri1966, 2020-04-02:
Current master (a6fe16c612a36a8d1786f7321087ef04f169e84f) contains a few changes which reduce the RAM used while processing the generated g-code.
This are the results of running this modified code on two different pcs:

```
[2020-04-02 08:18:59.470690] [0x00001628] [info]    Exporting G-code... WorkingSet: 506MB; PrivateBytes: 317MB; Pagefile(peak): 317MB(781MB)
[2020-04-02 08:18:59.484655] [0x00001a6c] [debug]   Loading print object toolpaths in parallel - start (GLVolumeCollection RAM: 2MB GPU: 0MB Both: 0MB) WorkingSet: 506MB; PrivateBytes: 319MB; Pagefile(peak): 319MB(781MB)
[2020-04-02 08:19:01.865856] [0x00001a6c] [debug]   Loading print object toolpaths in parallel - finalizing results (GLVolumeCollection RAM: 1,908MB GPU: 0MB Both: 0MB) WorkingSet: 2,414MB; PrivateBytes: 2,232MB; Pagefile(peak): 2,232MB(2,247MB)
[2020-04-02 08:19:34.620801] [0x00001a6c] [debug]   Loading print object toolpaths in parallel - end (GLVolumeCollection RAM: 1,908MB GPU: 1,905MB Both: 1,905MB) WorkingSet: 4,314MB; PrivateBytes: 4,171MB; Pagefile(peak): 4,171MB(4,171MB)
[2020-04-02 08:19:34.777046] [0x00001628] [trace]   Exported layer 0 print_z 0.2, time estimator memory: 0MB, analyzer memory: 0MB WorkingSet: 4,318MB; PrivateBytes: 4,171MB; Pagefile(peak): 4,171MB(4,171MB)
...
[2020-04-02 10:33:36.359745] [0x00001628] [trace]   Exported layer 1289 print_z 193.55, time estimator memory: 185MB, analyzer memory: 740MB WorkingSet: 3,069MB; PrivateBytes: 5,166MB; Pagefile(peak): 5,166MB(5,883MB)
[2020-04-02 10:33:39.583907] [0x00001628] [trace]   Exported layer 1290 print_z 193.7, time estimator memory: 185MB, analyzer memory: 740MB WorkingSet: 3,070MB; PrivateBytes: 5,166MB; Pagefile(peak): 5,166MB(5,883MB)
[2020-04-02 10:33:43.382875] [0x00001628] [trace]   Exported layer 1291 print_z 193.85, time estimator memory: 185MB, analyzer memory: 741MB WorkingSet: 3,070MB; PrivateBytes: 5,166MB; Pagefile(peak): 5,166MB(5,883MB)
[2020-04-02 10:33:43.945245] [0x00001628] [debug]   Time estimator post processing WorkingSet: 3,070MB; PrivateBytes: 5,166MB; Pagefile(peak): 5,166MB(5,883MB)
[2020-04-02 10:34:33.965906] [0x00001628] [debug]   Preparing G-code preview data WorkingSet: 3,070MB; PrivateBytes: 5,166MB; Pagefile(peak): 5,166MB(5,883MB)
[2020-04-02 10:34:39.591724] [0x00001628] [info]    Exporting G-code finished WorkingSet: 2,455MB; PrivateBytes: 4,489MB; Pagefile(peak): 4,489MB(5,883MB)
[2020-04-02 10:34:40.732092] [0x00001a6c] [debug]   Loading G-code extrusion paths - start (GLVolumeCollection RAM: 0MB GPU: 0MB Both: 0MB) WorkingSet: 1,160MB; PrivateBytes: 1,285MB; Pagefile(peak): 1,285MB(5,883MB)
[2020-04-02 10:34:40.763333] [0x00001a6c] [debug]   Loading G-code extrusion paths - create volumes (GLVolumeCollection RAM: 0MB GPU: 0MB Both: 0MB) WorkingSet: 1,160MB; PrivateBytes: 1,285MB; Pagefile(peak): 1,285MB(5,883MB)
[2020-04-02 10:34:40.794587] [0x00001a6c] [debug]   Loading G-code extrusion paths - populate volumes (GLVolumeCollection RAM: 16MB GPU: 0MB Both: 0MB) WorkingSet: 1,160MB; PrivateBytes: 1,294MB; Pagefile(peak): 1,294MB(5,883MB)
[2020-04-02 10:35:05.291009] [0x00001a6c] [debug]   Loading G-code extrusion paths - end (GLVolumeCollection RAM: 1,891MB GPU: 1,859MB Both: 1,859MB) WorkingSet: 4,639MB; PrivateBytes: 4,775MB; Pagefile(peak): 4,775MB(5,883MB)
```

```
[2020-04-02 08:12:42.664925] [0x000027cc] [info]    Exporting G-code... WorkingSet: 399MB; PrivateBytes: 383MB; Pagefile(peak): 383MB(733MB)
[2020-04-02 08:12:42.668914] [0x00000cb4] [debug]   Loading print object toolpaths in parallel - start (GLVolumeCollection RAM: 2MB GPU: 0MB Both: 0MB) WorkingSet: 399MB; PrivateBytes: 385MB; Pagefile(peak): 385MB(733MB)
[2020-04-02 08:12:43.545601] [0x00000cb4] [debug]   Loading print object toolpaths in parallel - finalizing results (GLVolumeCollection RAM: 1,908MB GPU: 0MB Both: 0MB) WorkingSet: 2,307MB; PrivateBytes: 2,299MB; Pagefile(peak): 2,299MB(2,338MB)
[2020-04-02 08:12:45.061333] [0x00000cb4] [debug]   Loading print object toolpaths in parallel - end (GLVolumeCollection RAM: 1,908MB GPU: 1,905MB Both: 1,905MB) WorkingSet: 2,311MB; PrivateBytes: 4,862MB; Pagefile(peak): 4,862MB(4,862MB)
[2020-04-02 08:12:45.132269] [0x000027cc] [trace]   Exported layer 0 print_z 0.2, time estimator memory: 0MB, analyzer memory: 0MB WorkingSet: 2,311MB; PrivateBytes: 4,862MB; Pagefile(peak): 4,862MB(4,862MB)
...
[2020-04-02 09:05:42.949038] [0x000027cc] [trace]   Exported layer 1289 print_z 193.55, time estimator memory: 185MB, analyzer memory: 740MB WorkingSet: 1,110MB; PrivateBytes: 5,852MB; Pagefile(peak): 5,852MB(6,568MB)
[2020-04-02 09:05:44.215745] [0x000027cc] [trace]   Exported layer 1290 print_z 193.7, time estimator memory: 185MB, analyzer memory: 740MB WorkingSet: 1,110MB; PrivateBytes: 5,852MB; Pagefile(peak): 5,852MB(6,569MB)
[2020-04-02 09:05:45.543373] [0x000027cc] [trace]   Exported layer 1291 print_z 193.85, time estimator memory: 185MB, analyzer memory: 741MB WorkingSet: 1,110MB; PrivateBytes: 5,852MB; Pagefile(peak): 5,852MB(6,569MB)
[2020-04-02 09:05:45.764749] [0x000027cc] [debug]   Time estimator post processing WorkingSet: 1,111MB; PrivateBytes: 5,852MB; Pagefile(peak): 5,852MB(6,569MB)
[2020-04-02 09:06:01.091258] [0x000027cc] [debug]   Preparing G-code preview data WorkingSet: 1,111MB; PrivateBytes: 5,852MB; Pagefile(peak): 5,852MB(6,569MB)
[... cut ...]
[2020-04-02 09:06:03.002544] [0x00000cb4] [debug]   Loading G-code extrusion paths - populate volumes (GLVolumeCollection RAM: 16MB GPU: 0MB Both: 0MB) WorkingSet: 317MB; PrivateBytes: 836MB; Pagefile(peak): 836MB(6,569MB)
[2020-04-02 09:06:07.029729] [0x00000cb4] [debug]   Loading G-code extrusion paths - end (GLVolumeCollection RAM: 1,891MB GPU: 1,859MB Both: 1,859MB) WorkingSet: 2,226MB; PrivateBytes: 4,925MB; Pagefile(peak): 4,925MB(6,569MB)
```

As can be seen, for this particular model, the RAM used by time estimator data and G-code analyser data moved from 3116MB to 926MB (70% reduction)

bubnikv, 2020-12-09:
Could you please try with PrusaSlicer 2.3.0-beta2?
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta2

rtyr, 2021-09-16:
No response. Should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/a6fe16c612a36a8d1786f7321087ef04f169e84f. Closing.

closed 2021-09-16 by rtyr