prusa3d/PrusaSlicer #3678: X-Ray mode on plater

neophyl, 2020-02-20:
### Version
N/A

* _Describe the problem_
The process of placing smaller modifiers that end up completely inside of the model (to avoid intersecting with perimeters as an example) can be difficult.
Would it be possible to implement an x-ray like mode so that you could see where they were.

The same mode could also be potentially helpful in detecting identical overlaid models as the shading could be more opaque where they overlap.

Cheers.

_Is this a new feature request?_
Yes

ekmek-civ, 2020-11-09:
Transparency or wire model or some way to look inside

spkane, 2025-02-21:
This is a surprisingly quiet ticket, given the number of duplicates that have been created over the years.

Does this just need someone to take it on? Or are there some concerns about how to even design or support such functionality?

cc/ @kubispe1 @JohnnyDeer

k20stitch, 2026-04-09:
Bump for visibility

Tupson444, 2026-09-13:
still relevant