prusa3d/PrusaSlicer #3680: Temperature compensation?

FlailAway, 2020-02-20:
### Version
All future

### Operating system type + version
All/Any

### Behavior
_Is this a new feature request?_
Yes

None of the firmware that I have used has ever offered temperature compensation for an off-reading Thermistor.

What about adding one into the **Filament** section? I had been having issues with clogging in the hotend and I finally attached real Temperature measurement. One K-Type to the hotend-heating block and another one to the lower fins on the E3D. 

To my surprise, the Thermistor is reading 15-C Low, so little wonder the heat was creeping up the E3D when printing at a required 260-C Nylon filament temp that Marlin was content with at an actual 275 as it appeared to be 260 and all is well with that little misguided world.

I now adjust all temps down as you can see, "190" in the Marlin Display and "203.1" for the real thing. It would be really handy to have a box near the Nozzle and Bed temps for a compensation value if required.

Thanks
p.s. Yes, I have validated the meter and both Thermocouples are within 0.5-C.

[image]

antimix, 2020-02-21:
- What about to correct the Thermistor table temp values in Marlin source code according with your thermistor type ?
Ideally the printer should be 100% calibrated, X,Y,Z movement distances and temperatures should be correct.

lukasmatena, 2020-03-19:
I personally don't think this is slicer's business. The printer and its firmware are responsible for temperature readings and that's where it should be fixed (e.g. by allowing some sort of calibration or a temperature offset). You can already workaround the problem by setting the filament temperature accordingly, I don't think we need another config value for this. Not to mention that after users start sharing such profiles, not knowing about this...

rtyr, 2020-04-18:
I agree with @lukasmatena.

Panayiotis-git, 2020-04-18:
15oC difference points to a wrong thermistor table selected at the Marlin.

This is the difference between the Marlin table 5 and table 13. At room temperature (25oC) they both report the same temperature. But at 200oC there is a 15oC difference. Make sure that you select the table 13.

If you have got the thermistor from a China supplier it is the most probable that they report a wrong thermistor type. I had the exact same problem with a thermistor I recently bought from TL (aliexpress).

rtyr, 2021-02-08:
As mentioned above, this issue should be fixed on the printer/fw side. Closing.

closed 2021-02-08 by rtyr