prusa3d/PrusaSlicer #3682: UI problem wiht perimeter >= 1mm

martinpirringer, 2020-02-20:
### Version
_Version of PrusaSlicer used goes here_
2.1.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
1.) Use a big nozzle
2.) Enter any perimeters that is >1mm the UI will ask a question to convert to % if yes is hit it will crash if no it sometimes saves sometimes does not Makes working with bigger nozzles (,8, 1, 1.2) very difficult
_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

FidelCapo, 2020-03-16:
Can you please retest it with PrusaSlicer 2.2.0-rc4? I think this issue should be fixed. Thank you.

rtyr, 2020-04-18:
Should be fixed. Closing. Feel free to reopen if problem persists.

closed 2020-04-18 by rtyr