prusa3d/PrusaSlicer #3683: Fix encoding in Hollowing

AndylgTom, 2020-02-20:
**Versions**
2.2.0-alpha4

**Operating system type + version**
windos10_64

**Behavior**
Pls, correct text encoding in Hollowing (Russian language).

[image]

bubnikv, 2020-02-21:
Is the localization of Russian correct in the other in-3d-scene dialogs as
the SLA supports and the Cut by plane dialog?

AndylgTom, 2020-02-21:
It is in these 3 lines that the problem is. In other places, everything is fine.

lukasmatena, 2020-02-21:
@AndylgTom Thanks for letting us know. It was hopefully fixed by 598ec046391c13ff587611f9fe6746b73b457ef3 for 2.2.0-beta. Could you please better retest after it is out (it will be shortly)?

AndylgTom, 2020-02-21:
Ok, of course.

AndylgTom, 2020-02-23:
Bug fixed, thanks.

lukasmatena, 2020-02-23:
Thanks for the confirmation.
Closing.

closed 2020-02-23 by lukasmatena