prusa3d/PrusaSlicer #3685: Add additional Placeholder variables?

felixekman, 2020-02-20:
### Version
_PrusaSlicer 2.2.0-alpha4_

I'd like to suggest the addition of a few Placeholder variables, most (all?) of which are already exposed in the code/processor, but not addressable in the "Start G-code" window.

**Some basic information about the print**
[filament_stats]
[print_time]
[normal_print_time]
[used_filament]
[total_used_filament]
[total_weight]
[extruded_volume]
[total_weight]

**For a thumbnail preview on other machines/boards/LCDs than the with Prusa Mini**
Index for jpeg/png(base64) and X/Y size
[thumnail_png_100x100]

**Values in mm for estimating work area movements of the machine**
[max_x]
[max_y]
[max_z]

Min should reasonably be able to take negative values (for outside of bed moves)
[min_x]
[min_y]
[min_z]

**Current Ref**
https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/Print.cpp
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

bubnikv, 2020-03-15:
please note that some variables are not yet known when processing the custom G-code sections. For example the print time cannot be known.

bepstein111, 2021-04-18:
I would also like to see this. In order to get the TFT** displays to show extra info at the end of a print, one needs to embed it via a `M118 P0 filament_data ` followed by W:[total_weight], and L:[total_used_filament].
Would be nice to embed this in my start/end gcode.

bepstein111, 2021-07-14:
The [Github for the TFT](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) wants:
M118 P0 filament_data L:{filament_amount}m
M118 P0 filament_data W:{filament_weight}g
M118 P0 filament_data C:{filament_cost}

But entering {extruded_volume} for the first variable returns an Invalid Data error, even though it's listed on the placeholders page.

closed 2026-09-29 by prusaslicer-bot[bot]