prusa3d/PrusaSlicer #3686: [2.2.0] Ignoring the Z speed settings on Delta

FlailAway, 2020-02-20:
### Version
2.1.1-linux-x64 appimage
2.2.0-Linux-64 appimage

### Operating system type + version
Mint 19.3 + Cinnamon

### 3D printer brand / version + firmware version (if known)
Anycubic Delta Linear Plus
MKSA-Gen-L
Marlin 1.1.9

### Behavior
I have set the Z in Machine limits, 
Maximum Feed Rate Z = 20mm/s
Maximum Acceleration Z = 200
Maximum Jerk Z = 2

This is the start of the GCode file
 generated by PrusaSlicer 2.2.0-alpha4+linux-x64 on 2020-02-20 at 20:06:06 UTC
; 
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M201 X3000 Y3000 Z200 E3000 ; sets maximum accelerations, mm/sec^2
M203 X130 Y130 Z20 E1000 ; sets maximum feedrates, mm/sec
M204 P1000 R1000 T1000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X4.00 Y4.00 Z2.00 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

G1 Z0.300 F3600.000
G1 E-4.00000 F2400.00000
G92 E0
G1 X-18.600 Y-36.048 F3600.000
G1 E4.00000 F2400.00000
G1 F1800.000

This is what is in the GCode file at Layer changes and it is too fast.
G1 Z0.300 F3600.000
G1 Z0.500 F3600.000
G1 Z0.700 F3600.000
G1 Z1.100 F3600.000
G1 Z0.900 F3600.000

I fully realize the Ultimaker doesn't have a Delta but Setup offers the option so it would be nice to see it slowing down on Z moves.

rtyr, 2020-02-20:
Your machine limits are not correct. You have to use same values for XYZ on delta printer (acceleration, jerk, max. feedrate).

ChrisRiddell, 2020-02-21:
I have this issue with a non Delta

I have been working on a Ultimaker 2+ Profile and the Z movement is just to fast.

```
M201 X1000 Y1000 Z400 E3000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z8 E80 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
```

GCode file at Layer changes and it's a little to fast.
```
G1 Z0.400 F7200.000
G1 Z0.550 F7200.000
```

example Cura `F600` and S3D `F1002`

rtyr, 2020-02-21:
It should not be a problem, maximum feedrate will be limited by firmware. F7200 will be limited to F480 in this case (M203 Z8).

FlailAway, 2020-02-21:
> Your machine limits are not correct. You have to use same values for XYZ on delta printer (acceleration, jerk, max. feedrate).

Thanks I understand that but those settings I show above were AFTER I had tried doing the "normal" way and having all XYZ set the same. It is using the TravelSpeed for all Z-moves.

Not a novice at this so have been trying to sort it out before posting here. In fact I wrote a rough and dirty C++ program to go through the GCode file and change all the lines with "Z" + "F4800" in the line to "F600"

Works, but not a good solution to have post process every dang GCode file.

FlailAway, 2020-02-21:
> It should not be a problem, feedrate will be limited by firmware. F7200 will be limited to F480 in this case (M203 Z8).

{grin} "should not" is not working. As mentioned in my reply above, my C++ program changes it to F600 and the problems it causes all go away with the modified GCode file.

FlailAway, 2020-02-21:
> I have this issue with a non Delta
I get a repeatable layer-shift with the delta because of the speed.
`Code causes a layer shift
G1 X32.834 Y-12.577 E241.82393
G1 Z1.100 F4800.000
G1 E237.82393 F2400.00000
G92 E0
G1 X-14.030 Y0.766 F4800.000
G1 E4.00000 F2400.00000
G1 F4800.000
G1 X-13.868 Y-0.511 E4.04355
`
The X-axis motor gives an hellacious burp-noise jumping teeth. If I change that 
`
G1 Z1.100 F4800.000
`
to
`
G1 Z1.100 F600.000
`
I think they requirement to move all 3-axis down for the Z and then a huge offset (G1 X-14.030) for the individual motors causes Marlin or maybe the TCM2208 or something to get confused.

rtyr, 2020-02-21:
"Should not" was answer to ChrisRiddell and I believe it was correct.

Unfortunately, if your printer cannot handle Z movement within set XYZ maximum feedrates, then you have almost certainly some different issue with your printer. In fact, similar issues can be caused by different acceleration/jerk values on each delta tower.

In this case, machine limit for Z axis means machine limit for one tower of delta printer, not for effector movement in Z direction like you probably want. There is no print setting to set Z feedrate independently atm (from toolhead/print perspective) and I would personally never use such setting for delta printer, where travel speed actually makes sense.

Don't take it wrong, I just think there is some different issue with your printer which should be sorted first.

FlailAway, 2020-02-21:
Thanks, but I dropped the speed JUST on that one line that kept giving the layer offset. I tried with four different speeds and when speed was at 600 it worked perfectly.

There is nothing wrong with the printer when it refuses to print only with that one GCode line unless it is set at 600mm/s to 720mm/s. I regularly print at speeds at 120mm/s and above with zero problems. 

Simply stated, PrusaSlicer should NOT be using the TravelSpeed for ANY printing moves, including Z. 

> "Should not" was answer to ChrisRiddell and I believe it was correct.
Perhaps in future you might like to consider using the "quote reply," option so people reading will understand just what you are referring to. I assume that you were referring to the OP -- me.

FlailAway, 2020-02-21:
In case you are wondering about the printer not being good enough, here it is right now powering along in draft-0.2 x 2-layers at 120mm/s on the same thing that PrusaSlicer shifted layer 1.10, but sliced with Cura. Sadly, I cannot wait until this is understood and fixed for delta printers in PrusaSlicer. Looks like a pretty OK printing machine to me.

[image]

ChrisRiddell, 2020-02-22:
My comment can be disregarded I have a hardware fault. and I misunderstood the original ticket.

rtyr, 2020-03-22:
I believe we can close this issue, since you basically want independent print setting for Z movement speed. Machine limits are not meant for this purpose.

There is already existing issue for that, https://github.com/prusa3d/PrusaSlicer/issues/2945. Closing here.

closed 2020-03-22 by rtyr