prusa3d/PrusaSlicer #3689: PrusaSlicer 2.1.1 locks up when opening on 2nd monitor in windows 10

ian99rt, 2020-02-21:
### Version
2.1.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version: Windows 10 Pro 64bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
PrusaSlicer opens fine on main monitor #1. If moved to Monitor #2, closed, and restarting Prusaslicer defaults to Monitor #2 it will hang and be nonresponsive. Had to duplicate display to move opening location of program to Monitor #1 and works again. 

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-02-22:
Isn't it the infamous Nahimic issue? 
https://github.com/prusa3d/PrusaSlicer/issues/2522
https://github.com/prusa3d/PrusaSlicer/issues/2142

rainylibrary, 2020-05-02:
I have the same/similar issue, only exacerbated since my original main display broke (thanks VR...) so I can no longer move PrusaSlicer back to that display to get it working again.

Win 10, 64
Asus Maximus XI Code
Twin RTX 2080TI, not in SLI, displays split between GPUs
PrusaSlicer v2.2.0

Additionally, I do not have sonic studio installed, and add/remove programs has nothing when I search for Nahimic or sonic. I do, however have the Nahimic service. But even when I stopped the service my display driver crashes. It seems like whatever PrusaSlicer is doing also causes nvlddmkm to crash, with the Nahimic service running and not. While PrusaSlicer does not crash for me since updating to 2.2.0, any attempt to rotate the build plate or click within the application causes the application to lag heavily (mouse cursor jumps from position to position) and eventually the display driver crashes again.

I seem to recall the last time I encountered this issue there was a file that could be deleted from the AppData/path/to/prusa that reset the application settings, which also fixes the issue.

kocikdav, 2021-02-01:
Hello. I would like to ask users having troubles with Nahimic libraries to try this build. It should display warning dialog about NahimicOSD.dll linked.
https://www.dropbox.com/s/on0dwea75gfx4cc/PrusaSlicer-2.3.0%2B127-win64-gf939cac84-202102011345.zip?dl=0

We know that this is just a warning rather than a solution, but currently we have no way to stop loading this library as it comes linked statically when we start our process.

Also if you are aware of another libraries of same source please let us know their names. You can find it with ProcessExplorer tool.

bubnikv, 2021-07-30:
There was no response. We believe the issue is the dreaded Nahimic driver injection.

Warning about the Nahimic DLL being injected into PrusaSlicer will be rolled out with PrusaSlicer 2.4.0-alpha1. It is not a "fix" for the nahimic issue, but at least the user will be informed.

closed 2021-07-30 by bubnikv