prusa3d/PrusaSlicer #3691: Using MMU the slicer

Tecd0c, 2020-02-22:
### Version
_2.2.0 Alpha 4 and standard 2.1.1_

### Operating system type + version
_Windows 10 x64_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.2.0-alpha4+win64
Build:     PrusaSlicer-2.2.0-alpha4+win64-202002091014

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,276MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.41
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
_Prusa MK3 upgraded S_

### Behavior
* _The slicer is slicing proper if I only select the MK3S as a printer once I chage it to MMU the slicing is not proper anymore,_

[image]
[image]
_this is not a new feature request_

#### Project File (.3MF) where problem occurs
[knight I_2x_each.zip](https://github.com/prusa3d/PrusaSlicer/files/4239360/knight.I_2x_each.zip)

bubnikv, 2020-02-22:
Would you please retest with today's 2.2.0-beta? Thanks.

Tecd0c, 2020-02-22:
Not exactly as the previous, but still not ok.

 

 

 

 

Stefan Merki                                   Stefan@Merki.Net <mailto:Stefan@Merki.Net> 

Robert-Koch-Str. 5                                         Tel.        02451-71031

52511 Geilenkirchen                                     Mobil    0178-4214876

                                                                               

ING-DiBa AG Frankfurt am Main               Steuer-Nr.  210 5106 0153

IBAN: DE24 5001 0517 5419 9994 08      USt- IdNr.   DE189480459

BIC: INGDDEFFXXX

bubnikv, 2020-02-22:
It looks good to me. Would you please point me as where the issue is now?

[image]

bubnikv, 2020-02-22:
I believe I understand your issue. The sign in the center of the shild is tiny and its geometry is very shallow therefore details are lost after slicing. The model will slice better if you scale it up.

Tecd0c, 2020-02-22:
Yes it is the roman letter I or II or III

But If I slice it normal without MMU setting it is doing it right.

So that’s strange.

 

 

Stefan Merki                                   Stefan@Merki.Net <mailto:Stefan@Merki.Net> 

Robert-Koch-Str. 5                                         Tel.        02451-71031

52511 Geilenkirchen                                     Mobil    0178-4214876

                                                                               

ING-DiBa AG Frankfurt am Main               Steuer-Nr.  210 5106 0153

IBAN: DE24 5001 0517 5419 9994 08      USt- IdNr.   DE189480459

BIC: INGDDEFFXXX

rtyr, 2021-02-08:
The details are to tiny to be printed with your extrusion width. 

> But If I slice it normal without MMU setting it is doing it right.

That's because the details (roman letters) doesn't have it's own perimeters in "single extrusion" mode.

I am closing this issue, the behavior is expected. You should either increase the size of the object, use smaller diameter nozzle or modify the design.

closed 2021-02-08 by rtyr