prusa3d/PrusaSlicer #3692: The skirt shouldn't cut the brim on the first layer

wavexx, 2020-02-22:
### Version
2.2.0-beta

### Behavior
In the current behavior (improved from #3516), the skirt splits the brim directly at the first layer to avoid overextrusion.

However, this is still _not_ ideal, since the brim will not effectively be as large as requested if a skirt is also used. Since the skirt creates a cut, the strong adhesion provided by the brim is limited up to the distance of the skirt (the remaining brim tears apart too easily).

See this example where the overhanging nature of the piece is forcing to have a skirt which is not aligned to the brim:

[image]

Please *also* note the gap visible on the inner left side between the skirt and brim, which is due to the fact that the skirt distance (2mm by default) is not a multiple of the extrusion width on the first layer

From the functional adhesion perspective, this behavior is also broken.

* _Expected Results_

The brim should have priority because it needs to be continuous to be effective.

On the first layer where both skirt and brim exist, and if the skirt overlaps the brim, I would expect the skirt to be printed _only_ on the build plate directly. To explain myself in other words, the skirt should be clipped by the area occupied by the brim.

On layers that follow, the skirt can be printed in full as usual, since the part that was covered with brim has the same height as the skirt that was printed on the bed.

This will also make it easier to print the first layer, since the skirt cutting the brim is causing a lot of discontinuities which make adhesion problematic with some materials.

BeatSlayer, 2020-02-22:
That's a feature, you can search the GitHub to find the discussion about it. You can also adjust the settings to solve the gap issues or just make it so the skirt ends out of the brim.

wavexx, 2020-02-22:
The discussion I've seen in #3516 is non-conclusive IMHO.

- Adjusting the skirt distance to a multiple doesn't remove the extra cuts and discontinuities introduced by overhanging  geometries as above. The brim should never be cut if you're looking of adhesion.
- Moving the skirt beyond the brim is not what you want as a heat shield. The skirt needs to be as close as possible.

The previous behavior (overprinting over the brim) was broken, but the current behavior is broken as well in my eyes.

golfromeo-fr, 2020-02-22:
>     * Moving the skirt beyond the brim is not what you want as a heat shield. The skirt needs to be as close as possible.

yes I have tested with ABS, it is a bit counter productive for a heat shield.

I think the skirt should be on top of the brim, otherwise the strength of the brim is broken

BeatSlayer, 2020-02-23:
Since we're at it, what I really believe it should be done is to restore the old skirt behavior and add an heat shield option that would print at a desired distance from the object and if a brim is selected, on top of the brim from layer 2 forward.

wavexx, 2020-02-23:
@BeatSlayer Is this equal to what I originally wrote or there's some extra difference? (maybe the old behavior was different than what I expected)

BeatSlayer, 2020-02-23:
The older skirt behavior placed the skirt outside of the brim by the distance set by the "skirt distance" parameter.
At the heat shield options we could even have the option to set it's height as a percentage or mm, so you could set an height of let's say 10mm if you only wanted a small heat shield for bed adhesion purposes or set it at like 100% if needed for ABS/PC/etc draft protection or ooze shield purposes.

misan, 2020-02-28:
I am surprised of this behaviour but I can understand why it might be preferred
[image]

On the other hand, I can see the brim lines do not stack on top of the skirt they are crossing so I would say the current behavior seems sane to me.

ChiefPoints, 2020-07-21:
I too was surprised to see the behavior of printing the brim over the top of the 1st layer of the skirt, I can only view this as broken because it kills a lot of the extra adhesion you get from using a brim. I'm currently manually placing the skirt well outside the brim footprint, but this should occur automatically. It only took one failed print to learn my lesson about expanding the skirt distance. I'd prefer to have the slicer automatically set the skirt distance from the brim via the skirt distance parameter. I don't think printing the skirt on top of the brim is a viable option either, as a huge benefit of the skirt is priming the nozzle. I'd rather not be priming the nozzle on brim lines I'm counting on for adhesion.

n8bot, 2020-07-21:
The reason they do this is to provide the extra support that the brim affords to the skirt as well.

Perhaps a compromise would be to "merge" the skirt and brim for layers where they overlap., but favoring the brim over the skirt. IE, on first layer, print the whole brim as normal, and only draw the skirt on areas where no brim exists, just a few lines here or there. Then, print the skirt directly ontop of the perfectly formed brim for the first layer where there is no brim.

ChiefPoints, 2020-07-22:
The main problem with the way it works currently is it forces the brim to print over the top of the skirt, and a solid skirt does you absolutely no good if your print fails because the brim was printed over the top of the skirt and didn't hold like it should. A single line of extrusion for the skirt doesn't have the same forces trying to lift it from the print bed, so doesn't require the brim to hold it down like the main print does. One option would be to print the brim first, and then forcibly print the skirt over the top of it, still not completely optimum, but much better than the current behavior. The purpose of using a skirt, whether it be to check initial flow, adhesion, or to build upon for a shield, should never overrule the purpose of the brim. 

> The reason they do this is to provide the extra support that the brim affords to the skirt as well.
> 
> Perhaps a compromise would be to "merge" the skirt and brim for layers where they overlap., but favoring the brim over the skirt. IE, on first layer, print the whole brim as normal, and only draw the skirt on areas where no brim exists, just a few lines here or there. Then, print the skirt directly ontop of the perfectly formed brim for the first layer where there is no brim.
> 
> [Edit: Err, though I guess the problem then arises that people use the skirt to purge, so it may not be fully formed, and if there are barely enough traces, it won't purge the extruder well. Perhaps the whole methodology of skirt and brim needs to be considered a non-purge thing, and purges are separate matters.]

cjshaker, 2021-04-12:
If you have a Brim configured, why do you also need an Object Outline? Doesn't a Brim fill that function as well?

lukasmatena, 2021-08-12:
Behaviour changed for upcoming 2.4.0-alpha1: https://github.com/prusa3d/PrusaSlicer/issues/4802#issuecomment-897492086
Closing.

closed 2021-08-12 by lukasmatena