prusa3d/PrusaSlicer #3694: Feature Request - Filename output format option to include "Scale"

heezus, 2020-02-22:
### Version
PrusaSlicer-2.1.1+-201912101508

### Operating system type + version
macOS Version 10.14.6 

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
- Slic3r supports putting the scale of the model in the output filename. I think supporting the same option would be very powerful for people needing to do repeat prints at various scales. This would be better for file management and be more verbose. See the following link and look at section "How can I specify a custom filename format for output G-code files?" https://github.com/slic3r/Slic3r/wiki/FAQ
- I think having the scale dumped into the gcode (printer configuration/comments at the end) would also be very helpful.

Marcel40625, 2021-03-12:
i would like to bump this 👍

RafaelSchridi, 2021-09-07:
This is currently available with [scale] 
[image]
But there is no way to only select one with [scale_0] like slic3r

Marcel40625, 2021-09-07:
> 
> 
> This is currently available with [scale]
> [image]
> But there is no way to only select one with [scale_0] like slic3r

With what Version? if i put it in print settings i dont get any filename so it breaks the generator ...

v2.3.3

RafaelSchridi, 2021-09-07:
I'm still on v2.3.1 but make sure you're not using curly brackets, for some reason they don't work.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]