prusa3d/PrusaSlicer #3695: another wxString type incompatibility

bnavigator, 2020-02-22:
Trying to package the new beta release, I encountered another wxString conversion error:

```
[ 1291s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-beta/src/slic3r/GUI/DoubleSlider.cpp: In member function 'wxString Slic3r::DoubleSlider::Control::get_tooltip(int)':
[ 1291s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-beta/src/slic3r/GUI/DoubleSlider.cpp:1010:59: error: operands to ?: have different types 'wxString' and 'const string' {aka 'const std::__cxx11::basic_string<char>'}
[ 1291s]  1010 |                     tick_code_it->gcode == ToolChangeCode ?
[ 1291s]       |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
[ 1291s]  1011 |                         from_u8((boost::format(_utf8(L("Extruder (tool) is changed to Extruder \"%1%\""))) %
[ 1291s]       |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 1291s]  1012 |                                                tick_code_it->extruder ).str())                                          :
[ 1291s]       |                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 1291s]  1013 |                         tick_code_it->gcode;
[ 1291s]       |                         ~~~~~~~~~~~~~~~~~~~                
[ 1291s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-beta/src/slic3r/GUI/DoubleSlider.cpp:1010:59: note:   and each type can be converted to the other
[ 1291s] make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:1112: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/DoubleSlider.cpp.o] Error 1
```

(Note: this is a very convoluted ternary operator expression) 
https://github.com/prusa3d/PrusaSlicer/blob/6ddb21fe067d1ec05cdab7d0701cd660ff6a5755/src/slic3r/GUI/DoubleSlider.cpp#L1002-L1013

(see also #3402 and commits 91cabe5, f60c117)

lukasmatena, 2020-02-25:
Could you please check if it was fixed by aa8c978? Thanks.

bnavigator, 2020-02-26:
I am still waiting on the [OpenSUSE build service](https://build.opensuse.org/package/show/home:bnavigator:branches:science/PrusaSlicer), but locally it built fine.

lukasmatena, 2020-02-27:
The unsafe conversions were disabled in PrusaSlicer cmake by 7c55674. It will hopefully fix this kind of issues for good.

I'm closing this issue, it should be fixed. Thank you for alerting us.

closed 2020-02-27 by lukasmatena