prusa3d/PrusaSlicer #3697: light green over light gray.. hardly readable

3Dietrich, 2020-02-22:
### Version
2.2.0-beta+

### Operating system type + version
MacOS Sierra 10.12.6

### 3D printer brand / version + firmware version (if known)
MK3S 

### Behavior
#### _Describe the problem_
maybe I'm color blind, but I can hardly read the light green texts (without zooming in the screen)

#### _Steps needed to reproduce the problem_
the green text is everywhere in the beta!

#### _Expected Results_
set the text color darker

#### _Actual Results_
missing contrast

#### _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
I can't work with that few contrast (without zoomin in the screen content)

#### _Is this a new feature request?_
I regard this as an bug

lukasmatena, 2020-02-22:
> I regard this as an bug

It probably is. Also the icons in the menu look funny - they are for the dark mode. Are you running in dark mode? What was the last PS version that you used and was fine (if any)? Was it 2.1.0 or did your try some of the alphas?

This is what I see on my Linux:
[image]

lukasmatena, 2020-02-25:
Ok, we managed to reproduce it on High Sierra 10.13. The problem is that macOS 10.12 and 10.14 allowed to enable "Dark Menu Bar & Dock Mode". If this is enabled, PrusaSlicer thinks that it is running in proper dark mode and sets white icons and light font colors, not knowing that the system will not provide the black background that it expects.

We should fix this, probably by checking macOS version and ignoring the dark mode status on anything older than Mojave 10.14, where actual dark mode was first introduced.

lukasmatena, 2020-02-28:
Fixed by https://github.com/prusa3d/PrusaSlicer/commit/7759eccf5624280ceeb2082ea43db964e446fab3 for 2.2.0-rc.

Closing.

closed 2020-02-28 by lukasmatena

3Dietrich, 2020-03-02:
2.2.0-rc
the colors and icons look great!