prusa3d/PrusaSlicer #3698: Unhandled Win32 Exception Error

ve3mic, 2020-02-22:
### Version
Ver. 2.2.0 beta Win32

### Operating system type + version
Windows 7 Professional SP1  32bit

### 3D printer brand / version + firmware version (if known)
Operating without a printer connected (patiently awaiting my Original Prusa Mini)

### Behavior
[image]

Problem occurs at random. Typically when changing to a different panel (navigating menus)

bubnikv, 2020-03-12:
Could you please capture a crash dump and provide it to us?
https://docs.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps

ve3mic, 2020-03-15:
Since updating to release candidate 2, I am unable to reproduce the issue.

ve3mic, 2020-03-16:
Problem reproduced by running the printer wizard after upgrading to RC4

ve3mic, 2020-03-16:
unable to attach crash dump file as it is too large (almost 200MB)

bubnikv, 2020-03-16:
could you provide the crash dump using some file sharing service? for example DropBox.

ve3mic, 2020-03-17:
Hi Vojtěch.
Please use this Google Drive link>  prusa-slicer.exe.8736.dmp

| 
| 
|  | 
prusa-slicer.exe.8736.dmp

 |

 |

 |

Best Regards,-Mike

bubnikv, 2020-03-17:
sorry, no google link drive provided. If this does not work (github may filter out the links which I believe it should not), you may send it to Vojtech Bubnik at prusa3d cz per e-mail.

bubnikv, 2020-03-17:
What prusaslicer version exactly was this crash dump collected for?

It seems to be crashing inside the intel graphics card driver, you may try to update that.
Also I see from the crash dump that you have dropbox integrated into PrusaSlicer running one thread, and I see also that you have google drive DLLs loaded. God knows whether they may cause havoc.

I may find out more if I can match the crash dump to the correct debug symbols. I tried 2.0.0-rc4, but they don't seem to match.

bubnikv, 2020-03-17:
Was it a 32bit or 64bit of PrusaSlicer that you captured your crash dump with? From the path of the binary it looks like it was a 32bit build, but the debugger seemed happy with a 64bit build.

ve3mic, 2020-03-17:
Hi Again Vojtech,
Sorry for the delay in getting back to you as I am working from home and have low internet bandwidth.I have captured another crash dump>  prusa-slicer.exe.5948.dmp

| 
| 
|  | 
prusa-slicer.exe.5948.dmp

 |

 |

 |

I terminated as many services & background processes beforehand, but I am able to reproduce the crash every time by running the Configuration Wizard. Once I click "Finish" the crash occurs. It is the 32bit version> PrusaSlicer-2.2.0-rc4+win32-202003151125
Regards,-Mike

bubnikv, 2020-03-18:
So here is the call stack where it crashes in your first dump that you managed to send me per e-mail.

[image]

@enricoturri1966 what do you think?

bubnikv, 2020-03-18:
Would you please try to update your graphics card driver?
https://www.intel.com/content/www/us/en/support/detect.html

bubnikv, 2020-03-18:
I wonder what would happen if you delete your your PrusaSlicer directory?

https://help.prusa3d.com/en/article/faq-prusaslicer_1789/

Q: Where are user profiles stored?
Windows C:\Users\username\AppData\Roaming\PrusaSlicer
Mac “HDD” ▸ ⁨Users⁩ ▸ ⁨username⁩ ▸ ⁨Library⁩ ▸ ⁨Application Support⁩ ▸ ⁨PrusaSlicer
Linux ~/.PrusaSlicer

and then install no system printer, but a custom one. It will render a simplified print bed and this piece of code where your machine is crashing is not called.

bubnikv, 2020-03-18:
Does PrusaSlicer 2.2.1 work for you on the same compute?

closed 2021-09-22 by bubnikv