prusa3d/PrusaSlicer #3699: Larger FOV when perspective view is enabled would be greatly appreciated (Perhaps even ability to change it)

NameTakenWasTaken, 2020-02-23:
### Version
2.1.1

### Windows 10 1909

### Behavior
*Perspective view very low FOV; Appears like orthographic
* Change FOV value for perspective view
* _Actual Results_ More life like view of the print in software
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ Yes.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
****
[image]

jmd73, 2020-02-24:
You can zoom in/out with the scroll wheel of your mouse or with the i and o keys.

NameTakenWasTaken, 2020-02-26:
> You can zoom in/out with the scroll wheel of your mouse or with the i and o keys.

Yes but you cannot move the camera physically closer, which means that to get close you have to zoom the FOV super far

neophyl, 2020-02-26:
I'm struggling to understand what you mean.  Do you have an actual example of some software with clear images to illustrate what you want.
as far as i know zooming is the virtual equivalent of moving the camera closer, as we can pan, rotate and zoom already I have never seen any other software that can do something different, even using something like a spacemouse for control.

NameTakenWasTaken, 2020-02-26:
Blender is a great example. When you scroll in Blender the FOV doesn't
change, only the cameras position changes.
FOV merely makes the object look closer because it takes up more of your
screen.

neophyl, 2020-02-26:
mmm I use blender 2.79b all the time and I've not noticed anything like that. It behaves the same as far as I can see.

bubnikv, 2020-02-26:
The perspective camera in PrusaSlicer changes focal length with zoom level,
which is quite unusual. We will discuss internally whether it is worth
changing as there is only one customer complaining about it.

simon-budig, 2021-03-28:
FWIW: Here is a 2nd customer complaining about this issue. It took me a while to realize that the "perspective view" toggle in the config actually does anything and I assumed that it was just nonfunctional...

NHodgesVFX, 2021-10-17:
I would also like this feature right now orthographic and perspective look very similar

n8bot, 2021-11-24:
Just for reference, Fusion 360's render environment moves the camera with the scroll wheel. But, that's besides the point. I think the current behaviour can be improved enough to be acceptable for everyone. See the above mentioned PR.

AngryCarrot789, 2024-05-31:
Will this feature ever be added? The perspective camera might as well be called orthographic, it's horrible to look at and I can barely tell what's what on my models

Zerreth, 2025-01-09:
Had to look this up since it was driving me insane. And yes the perspective view has horrible fov because it is used to "zoom in" as opposed to just moving the camera closer along the view direction. It is the equivalent of not using dTime whilst programming and thus it's a stupid implementation. Camera Distance ≠ Camera FOV. Those are 2 totally different things. Literally all 3D software I've worked with in the last 25 years have not made this mistake. ( SoftImage, 3ds Max, Maya, Zbrush, Mudbox, Blender, id Tech 2.5, GoldSrc, Source, Unreal, Unity, ... )

araghon007, 2025-01-28:
I started using Prusa Slicer recently and immediately noticed the ortographic camera... Except it wasn't ortographic, it was perspective! Just with a very, very low field of view. Every single piece of 3D software and slicers I've used in the past always move the camera with scrol wheel in perspective view, instead of zooming in.

TheSwaGiraffe, 2025-06-18:
Scroll wheel to actually move the camera position would be nice because it would allow people to navigate inside of holes inside their prints. I was trying to switch to perspective so I could see inside of my print until I realized that it already was in perspective mode. I think a setting should be added similar to FreeCAD where you can choose between other engine's navigation systems like blender or Tinkercad or other 3d software.

araghon007, 2026-09-02:
Still very much an issue, k thx

Again, what's even the point of having a perspective projection if it looks 99% like orthographic anyway