prusa3d/PrusaSlicer #3701: [SLA] supports intersect with other objects

acorderob, 2020-02-23:
### Version
2.2.0 beta

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
* _Describe the problem_
When trying to print several objects on the plate, specially when a small one is positioned under a bigger one, the bigger one supports intersect with the small one.
* _Expected Results_
The supports should avoid the small object or start in its surface.
* _Actual Results_
[image]

_Is this a new feature request?_
No

tamasmeszaros, 2020-03-02:
This is just how PrusaSlicer works at the moment. It prepares each object independently.

bubnikv, 2020-03-02:
The reason for this behavior is speed and flexibility of the workflow.
AFAIK other SLA slicers work the same way. We may try to improve upon the
status quo, but it will not be quick and easy.

closed 2020-03-16 by tamasmeszaros