prusa3d/PrusaSlicer #3703: Issue when using multiple extruders

TheEagle13, 2020-02-23:
### Version
2.1.0
### Operating system type + version
Fedora 31

### 3D printer brand / version + firmware version (if known)
Formbot T-Rex 3
### Behavior
Hey all, I've found an issue with the slicer on a model with multiple parts and some open areas/gaps.

When I set everything to the same extruder, the preview is accurate, however, when I set the second part to the second extruder, it starts to fill in parts of the gaps randomly. (See photos) The gaps are 2-3mm at the smallest, so they shouldn't qualify as "gaps" that should be filled. Any ideas about what's going on, or how to fix it?

Correct when using 1 extruder:
[image]

Incorrect when using 2 extruders:
[image]

#### Project File (.3MF) where problem occurs
[floor_stand.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4242455/floor_stand.3mf.zip)

bubnikv, 2020-02-24:
I cannot reproduce with PrusaSlicer 2.2.0-beta
Would you please test with PrusaSlicer 2.2.0-beta?

TheEagle13, 2020-02-24:
It happens the same with the 2.2.0-beta Linux AppImage for me

rtyr, 2020-05-18:
The model seems broken. Is it OK with this 3MF (repaired model)?

[floor_stand_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/4642878/floor_stand_fixed.zip)

TheEagle13, 2020-05-18:
Doesn't appear broken to me. But my model does seem to be working fine in the latest release of Prusa Slicer, so I'm guessing the bug/regression was fixed.

closed 2020-05-18 by TheEagle13