prusa3d/PrusaSlicer #3704: [Feature Request] Support for multiple fans for IDEX extruders

TheEagle13, 2020-02-23:
### Version
2.1.0

### Operating system type + version
Fedora 31

### 3D printer brand / version + firmware version (if known)
Formbot T-Rex 3

### Behavior
The Formbot T-Rex3 Is a dual-IDEX printer with a separate controller for the fan on each print head. Currently, when printing with both extruders, the fan is never enabled for the 2nd print head. The fan speed is always attributed to the 1st print head.

BeatSlayer, 2020-02-24:
Can you turn it on by manually sending the Gcode? Does it stay on when toolchanging?
Could be a firmware bug.

TheEagle13, 2020-02-24:
It's not a firmware bug. PrusaSlicer doesn't generate the necessary Gcode for a secondary fan. It only generates code as if there is only one fan.

In all of my Gcode produced by PrusaSlicer it is M106 S<Speed>. IE:
M106 255

In order to control the speed of the second fan there needs to be a P<index> in the command too M106 P<index> S<speed>. IE:
M106 P1 S255

Same thing with turning the fans off, you need M107 P<index>

If I manually send the commands to the printer via octoprint terminal, everything works as expected. The issue is that PrusaSlicer doesn't seem to have an option to account for multiple fans

TheEagle13, 2020-05-18:
any update, or any "cheat code" that I could use to make this happen?

pkucmus, 2020-09-13:
I'm interested as well. On a Duet (RRF) when you setup different fans as "fan1" and "fan2" you need to do `M106 P1 Sxxx` and `M106 P2 Sxxx` to control the fans properly.

Kachidoki2807, 2020-09-24:
Any news on this topic?
I am currently building an i3 based IDEX machine, but indeed the FANs are not properly handled by PrusaSlicer.

Maybe someone has an idea to workaround this, using the custom G-Code sections and some magic placeholders (I wasn't able to find an appropriate one)?

TheEagle13, 2020-09-24:
As a temporary "workaround" I just add the following to my Start G-Code:
`M106 P1 S200`

Granted it's a static temperature for the whole print, but it's better than nothing. (Especially when you do a lot of remote printing.

TheEagle13, 2020-09-24:
I've also read that you can change the firmware to control both fans as one in the same, so if you're building your own, that may be an option until they have software allowing the control of 2 seperate fans

pkucmus, 2020-09-25:
That may not be a bad idea, in the "Tool change G-code" we could do something like:
```
{if layer_num == 3}  ; to start cooling after 2 layer
{if next_extruder == 0}
M106 P0 S1.0
M106 P1 S0
{elif next_extruder == 1}
M106 P0 S0
M106 P1 S1.0
{endif}
{endif}
```
And disable fan control in the filament settings.

ogland, 2020-10-11:
I very much want this too.

Kachidoki2807, 2020-11-10:
Hi,
I have now my MakerFr i3-RDX built in alpha revision (IDEX printer). I am currently tuning the slicer profile, but I am still stuck on the FAN speed that is not properly handled by PrusaSlicer, and apparently neither by Slic3r: https://github.com/slic3r/Slic3r/issues/4694

I found also this related topic: https://github.com/prusa3d/PrusaSlicer/issues/3583 

Generally speaking, the M106 handling should be reworked to fulfill at least the [P<fan_index>] parameter.
If it imply too much impact into the current implementation, PrusaSlicer could provide at least something like a "current_extruder_fan_speed" placeholder to be able to handle it from the toolchange custom G-Code.

ogland, 2020-11-10:
What I found is that with fanmux, the fan targeted with M106 without the P should be current extruder, but with T parameter in toolchange gcode, the M106 command fires BEFORE the toolchange and possibly only when the fan speed should change. If instead the fan speed could be set after toolchange gcode and always to the correct current speed I believe it would work good.

Also an idle fan speed is good to have with ooze stoppers and that would be very easy to enable with a tool change gcode in that case.

Kachidoki2807, 2020-12-16:
Hello,

I digged a bit into the code (it is very new to me, I am not used with the Prusa workflow). I found that there is potentially an inversion like mentioned by @ogland.
https://github.com/prusa3d/PrusaSlicer/blob/9fdf8c8b8d46e26968baa5d6e6b80bfcea046e0c/src/libslic3r/GCode/CoolingBuffer.cpp#L735-L741
My guess is that the line 741 should be before the if statement (move to 737) to place the M106 **after** the Tx because it is the fan speed of the new extruder that should be applied here.

I built and tested this small fix with success.

andrewseago, 2021-06-11:
Any update on this? I'm super interested in it. One way could be to add a setting per extruder for the parts fan number.

schuessm, 2021-12-09:
I don't understood the problem, normaly it will be done by the firmware. I use Klipper and there is a little bit code in configuration like this

[gcode_macro M106]   
gcode:
    {% if printer.toolhead.extruder is defined %}
      {% if params.S is defined %}
        SET_FAN_SPEED FAN=fan_part_{printer.toolhead.extruder} SPEED={params.S|int * 0.003921569}    #*0.003921569
      {% else %}
        SET_FAN_SPEED FAN=fan_part_{printer.toolhead.extruder} SPEED=1.0    #255
      {% endif %}
    {% else %}
      {% if params.S is defined %}
      SET_FAN_SPEED FAN=fan_part_extruder SPEED={params.S|int * 0.003921569} 
      SET_FAN_SPEED FAN=fan_part_extruder1 SPEED={params.S|int * 0.003921569} 
    {% else %}
      SET_FAN_SPEED FAN=fan_part_extruder SPEED=1.0
      SET_FAN_SPEED FAN=fan_part_extruder1 SPEED=1.0
      {% endif %}
    {% endif %}

[gcode_macro M107]
gcode:
    {% if printer.toolhead.extruder is defined %}
      SET_FAN_SPEED FAN=fan_part_{printer.toolhead.extruder} SPEED=0.0
    {% else %}
      SET_FAN_SPEED FAN=fan_part_extruder SPEED=0.0
      SET_FAN_SPEED FAN=fan_part_extruder1 SPEED=0.0
    
    {% endif %}

TheEagle13, 2021-12-09:
@schuessm
The problem is that modern more advanced idex printers now allow you to control the fan speed per head. Say for instance you are printing with something on head one that needs 100% fan, but the other head needs 0% fan.

So no, it's not a firmware thing (at least not with marlin), it's a thing that should be controlled by the slicer depending on the material/active head.

schuessm, 2021-12-09:
oh, sorry than I missunterstood the problem

TheEagle13, 2021-12-09:
PrusaSlicer works for Idex in general. I found you have to preheat both heads before starting the print, but I always preheat anyway, so it never really was a a problem for me.

But the problem in this issue is for different fan speeds per head.

schuessm, 2021-12-09:
How did you realize the prehead function? I didn't found a function? Which version of Prusa are you using

bubnikv, 2021-12-11:
@We at PrusaResearch did not manufacture IDEX or tool changer machines before. Now with the introduction of the tool changer Prusa XL printer, we are actively looking into the IDEX / tool changer requirements.

The topic we are looking into is how much of the tool changer support we shall put into firmware vs. into slicer generated tool change G-code blocks. On our newer machines we use the stock Marlin.

BTW what firmware are you running? I suppose Marlin and RepRapFirmware on Wifi Duet, right?

themacboy, 2022-07-28:
I had the same problem. I'm managing it using macro script in Klipper... but that's not the best solution. It would be great if PS you could manage more than one tool head with a separate fan.

Thanks in advance.

Elevatea, 2023-06-04:
Just to add a possible solution, I have used the Post Processing Scripts from PS to automatically modify the gcode while saving it, to call a script that adds a second line that turns on the second fan every time it founds a M106 S255 (for the first fan). 

It's the python script below, all what you have to do is to save it as a python file, and then put the path to it with the "python" or "python3" command in the Post Processing Scripts.

`import sys

def modify_gcode(input_file):
    # Open the input file and read all lines into a list
    with open(input_file, 'r') as f:
        lines = f.readlines()

    # Open the input file again in write mode, this time we will overwrite it
    with open(input_file, 'w') as f:
        # Iterate over all lines in the file
        for line in lines:
            # If the line is the fan control command for the first fan
            if line.strip() == "M106 S255":
                # Write the fan control command for the first and second fan each on a new line
                f.write("M106 S255\nM106 P2 S255\n")
            else:
                # If this line is not a fan control command, write it back to the file as is
                f.write(line)

# Call the function with the name of the file PrusaSlicer passed to the script
modify_gcode(sys.argv[1])
`

Here's how it works: This Python script reads in the G-code file created by PrusaSlicer, which is passed as a command-line argument to the script. The script checks each line of the G-code for the command "M106 S255", which turns on the first fan at full speed. Whenever it finds this command, it adds a new line with the command "M106 P2 S255" right after, turning on the second fan at full speed as well. This modified G-code file is then saved, ready to be used for 3D printing.

To use this script, save it to a .py file on your system. Then, in PrusaSlicer, go to Print Settings -> Output options -> Post-processing scripts and enter the path to the script, preceded by the command to run a Python script, which is typically either "python" or "python3". The exact command and path will depend on your specific system setup. For example, it might look like this: python3 /path/to/your/script/modify_fans.py.

Calypsoblur, 2023-09-25:
I haven't tested it yet on my Snapmaker J1S, but I have updated the current (2.6.1) code to add the P in the fan speed setting and the current extruder. Have not added to M107. It does compile, and put out g code as expected (calling out Px when ever M106 is called). The code is only for Marlin or similar gcode flavor, which should include RepRap and Klipper, but as stated in first sentence, not tested. This will require you to compile your own version. Include my files at your own risk. I have added "by CRobson" on all lines I modified. 
PrusaSlicer/src/libslic3r/GCode.cpp
PrusaSlicer/src/libslic3r/GCodeWriter.cpp
PrusaSlicer/src/libslic3r/GCodeWriter.hpp
PrusaSlicer/src/libslic3r/GCode/CoolingBuffer.cpp
PrusaSlicer/src/libslic3r/GCode/CoolingBuffer.hpp
I don't remember making any changes to CoolingBuffer.hpp, and I can't see anything I may have changed, but I did mark it. 

you can download from my github
[Multi-Extruder Fan handling](https://github.com/Calypsoblur/PrusaSlicer/tree/main#prusaslicer)

Drizzt321, 2024-03-17:
I'm finding this is an issue with the [FormBot Marathon IDEX](https://www.vivedino.com/products/presale-marathon-fully-enclosed-independent-dual-extruder-idex-3d-printer-with-klipper-firmware) as well. When switching tool heads, PrusaSlicer doesn't have the Part Fan for either tool head turn on. 

Under `Filament Settings` -> `Cooling`, I have `Keep fan always on` checked. This is PrusaSlicer 2.7.2, PLA.

I haven't tried the script from @Elevatea yet, just found this ticket and wanted to chime in that this is very much needed.

YILI245, 2024-05-20:
> I'm finding this is an issue with the [FormBot Marathon IDEX](https://www.vivedino.com/products/presale-marathon-fully-enclosed-independent-dual-extruder-idex-3d-printer-with-klipper-firmware) as [...]

Hi,
I have the same problem in my Marathon.
Try to remove following code in after layer change:
{ifcurrent_extruder==0}
SET_FANSPEEDFAN=Tool_1_partfanSPEED=O.O
{else}
SET_FANSPEEDFAN=Tool_0_partfanSPEED=O.O
{endif}

And add following to the Tool change:
{if layer_num>=disable_fan_first_layers[0]}
 {if current_extruder==0}
    SET_FAN_SPEED FAN=Tool_0_partfan SPEED=0.0 ; 关闭0号挤出头的风扇
    SET_FAN_SPEED FAN=Tool_1_partfan SPEED=1.0 ; 开启1号挤出头的风扇
 {else}
    SET_FAN_SPEED FAN=Tool_1_partfan SPEED=0.0 ; 关闭1号挤出头的风扇
    SET_FAN_SPEED FAN=Tool_0_partfan SPEED=1.0 ; 开启0号挤出头的风扇
 {endif}
{else}
{endif}

marczis, 2024-09-21:
Any news on this? Does Pursa generate G code for multiple head (IDEX) setups?

jestepp, 2025-02-02:
Here for the same

AcesSoon, 2025-10-06:
I use Ai to write the post_processing scripts python which can utilise the Prusa Slicer .gcode, 
i'm using Flashforge Creator 3 Pro, the post_processing scripts is get currently M108 Tx (which is tool change .gcode) , when meet M106 line, add Tx on it. When switching tools(M108) , apply lastest fan speed.

If you're Px or another M code, make sure changging to suite your machine.

https://github.com/aces0605/Prusa_Slicer_Script/blob/main/path_fan_speed.py
[image]
Here how to use it.

saga111a, 2026-01-19:
@AcesSoon thank you! Looks like solution

Drizzt321, 2026-09-02:
Please keep this open, I'm willing to re-check once you release to stable/flatpak. IMO being in alpha isn't sufficient to label this legacy.