prusa3d/PrusaSlicer #3708: Thin Wall not sliced

muellerbru, 2020-02-24:
### Version
PrucaSlicer-2.1.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.1.1+win64
Build:     PrusaSlicer-2.1.1+win64-201912101512

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 34,298MB
OpenGL installation
GL version:   4.6.0 NVIDIA 442.19
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 970/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 340 lines cut ...]
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Original Prusa I3 MK3S/MMU2S
Firmware 3.8.1-2869

### Behavior
* _Describe the problem_
RC-Plane from 3dLabPrint (Piper Pawnee) -> Fuselage STL (Dusty fuselage 9.stl) import to Prusa Slicer -> the thin walls are not Reconized. I already checked my Konfigruations with Prusa Support (Sarah), and she got's the same, wrong result. My opinion: no matter how thin a wall is, with "Thin Wall dedectet" the Slicer should reconize this wall an at least sliced it with 1 Perimeter. the STL ist valid an it works in Simplify3d. 
The other problem ist, that the deliverd GCODE (and also the Sympli3d) is not valid for the new Slicer. Perhaps the problem is the MMU? And Sympli3d does also not support MMU.

* _Steps needed to reproduce the problem_
Load this STL and create Gcode in Slicer. No Matter which option (Thin wall, Extra Perimeter if needed etc.). and the Thin Walls are not generatet.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ No, I think it is a Bug
[Support.zip](https://github.com/prusa3d/PrusaSlicer/files/4246799/Support.zip)

Attached a Zip, contains: my 3mf, the original 3DLabprint, Piperpawnee STL, and the original 3DLabprint gcode, Original Factory file, Screenshot bevor and after Slicing, and a screenshot from Simplify3d after Slicing

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2020-02-24:
I think this is a duplicate of 3062
https://github.com/prusa3d/PrusaSlicer/issues/3062

muellerbru, 2020-02-24:
Yes I think also, but there is also a new Issue that I assume depends on MMU is not well supportet too on Simplify3d, and in the Gcode from 3DLabprint.

pstasek, 2020-04-08:
The MMU issue could be solved by manually loading the filament through LCD or adding an appropriate starting script to the stock Gcode or Simplify3D factory file. Just copy the starting script from Prusa Slicer settings.

bubnikv, 2021-06-01:
Duplicate of #3062

closed 2021-06-01 by bubnikv