prusa3d/PrusaSlicer #3709: External perimeter width with part modifier not behaving correctly

belgiancoder, 2020-02-24:
### Version
2.1.0 and 2.1.1

### Operating system type + version
macOS Catalina

### 3D printer brand / version + firmware version (if known)
MK3

### Behavior
* part modifier > extrusion width > external perimeter > 0.7 doesn't set the external perimeter width to 0.7 (I put 1.2 in the snapshots so you can see it)
* _Expected Results:_ The width changes 
* _Actual Results_: It doesn't, but it seams to change the internal perimeter
  * Without part modifier:
[image]
  * With part modifier:
[image]
[image]
[image]

rtyr, 2020-05-21:
Hi. Is this issue still present in 2.2.0? I am unable to reproduce it on similar test model. Can you provide your 3MF project file?

rtyr, 2021-02-08:
No response. Closing.

closed 2021-02-08 by rtyr