prusa3d/PrusaSlicer #3711: Regression: Empty layers when models are sliced

leftspin2k, 2020-02-25:
### Version
PrusaSlicer-2.2.0-beta Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Prusa MK3, stock

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Slice the model. 

[image]

[image]

[image]

[image]

This is a regression. Slice worked properly on Slic3r-1.42.0-beta+win64 as shown below:

[image]

[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other 
[Slantini_Peak_Slicing_Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/4247568/Slantini_Peak_Slicing_Bug.zip)
files need to be zipped first (.zip, .gz)_

leftspin2k, 2020-02-25:
Also worked in PrusaSlicer-2.1.1+win64
[image]

espr14, 2020-04-14:
There are issues in STL file. Use PrusaSlicer 2.2.0. Right click on the object and select "Fix through the Netfabb". Slicing will finish without error (even without fix).

closed 2026-09-29 by prusaslicer-bot[bot]