prusa3d/PrusaSlicer #3712: first layer of vase (after bottom layers) is bulging (0.6nozzle, 0.4 layerhight)

3Dietrich, 2020-02-25:
### Version
2.1.1+

### Operating system type + version
MacOS 10.12.6 Sierra

### 3D printer brand / version + firmware version (if known)
mk3s 3.9.0-3175 (with hemera tweak, but that's unrelated to the bug)

### Behavior
#### _Describe the problem_
The first layer of vase layers (after filling the bottom) seems too thick/bulging.
#### _Steps needed to reproduce the problem_
I used 0.6 nozzle, with the  template "0.40 Draft..." (Print Settings) and "Generic PLA" (Filament Settings) and switched to Vase mode. (I changed some other things too, but that's unrelated)
In doubt I suggest to load the .3mf and print the first layers (with a 0.6 nozzle!) to test - I'm convinced it's a (little) bug.

#### _Expected Results_
clean surface
#### _Actual Results_
watch pictures ([zip](https://github.com/prusa3d/PrusaSlicer/files/4247718/VaseModeFirstLayerOverBottomTooStrong.zip))

_Is this a new feature request?_
nope

#### Project File (.3MF) where problem occurs
[several photos and the .3mf and .gcode files in this zip file:
VaseModeFirstLayerOverBottomTooStrong.zip](https://github.com/prusa3d/PrusaSlicer/files/4247718/VaseModeFirstLayerOverBottomTooStrong.zip)

wavexx, 2020-02-25:
Duplicate of #2795

bubnikv, 2020-12-09:
Linear tapering of extrusion rate at the 1st spiral vase layer implemented with 298b730589371d39fe4a85851a9c72f87fc22d8e

closed 2020-12-09 by bubnikv

bubnikv, 2020-12-09:
Follow-up with 298b730589371d39fe4a85851a9c72f87fc22d8e
    Linear tapering could only be implemented for relative extruder distances.
    For absolute extruder distances, all the G1 Exxx in all spiral vase layers
    would have to be adjusted, which is much more complex and risky to do now
    before release.