prusa3d/PrusaSlicer #3716: Print Pause

gwak64, 2020-02-25:
### Version
2.20 beta Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win 10 Pro 64bit

### Prusa I3 MK3S Firmw 3.81

### Behavior
If there is a print pause to insert a magnet, no short message appears on display. Likewise, resume printing does not work. The printer starts but immediately pauses again.

[ED 363 2.zip](https://github.com/prusa3d/PrusaSlicer/files/4250576/ED.363.2.zip)

lukasmatena, 2020-02-25:
This is firmware-related issue that is known and has been already fixed. See https://github.com/prusa3d/PrusaSlicer/issues/3557

Closing. In case the problem does not disappear after firmware update, let us know. Thanks.

closed 2020-02-25 by lukasmatena

gwak64, 2020-02-25:
Wow, this was fast!

Thank you very much

Gerd

Burn after printing