prusa3d/PrusaSlicer #3717: MMU2 Purge tower, head stationary during filament change.

beegmouse, 2020-02-25:
### Version
2.10

### Operating system type + version
WIN10 x64

### 3D printer brand / version + firmware version (if known)
i3 MK3S + MMU2

### Behavior
During a filament change, there is a period of deliberate slow movement while the head changes temperature to match the new required temperature.
The head stops moving while a filament change happens and then the purge move takes place.

The problem is when the head remains static with cooling fans off the filament is changing from 215c BVOH/PLA to 230/240 PETG the head boils the previously extruded BVOH/PLA in the purge block. This causes a hard lump to appear in the purge block around where it stopped.

If the filament change takes multiple attempts or a user intervention is required the lump gets larger.
This can and has caused multiple non recoverable print failures for me.

My suggestion is that the head keeps moving slow during the change or the fan left on to keep the deposited filament from overheating, or if not possible, raise the head off the purge block, or out to one side.

doug62, 2020-02-29:
I have similar issues but poor IR/FINDA design/software sensing keeps creating pea sized balls here and then fails.  When i check sensors the sensors indicate the state (filament loaded and in extruder) but it just keeps pushing.

beegmouse, 2020-04-08:
That's not the same issue.
It sounds like your pinda optoswitch is not triggering early enough. try recalibrating it so that it only just comes on with your actual filament.
If your filament has a pointy tip and it only triggers the pinda at the widest part? There will be more material in the extruder than the printer expects.

(Tip I used, load filament into a cold head by hand. Watch the sensor input and squeeze the extruder gears so the filament is crushed a little. if your sensor output varies, you are too sensetive to deviations in diameter.)

But, fixing my problem would still help yours.
maybe they move this position around the block each layer so that the small over extrusions don't stack into big ones.

lukasmatena, 2020-04-08:
> During a filament change, there is a period of deliberate slow movement while the head changes temperature to match the new required temperature.
The head stops moving while a filament change happens and then the purge move takes place.

I don't think that there is any pause to allow changing temperature. It sound to me that you're describing the pause during the actual filament exchange by MMU2, which is difficult to get rid of in PrusaSlicer, because it is processing Tn command at that time and no other gcode can be processed before it is done.

beegmouse, 2020-04-08:
If that is uncontrollable, could the position of the pause/change shift couple of mm every layer?

nsensfel, 2023-05-12:
Would it be possible to move the head to a position right next to the wipe tower just before sending the `Tx` command to the MMU? By having it hang above nothing, this blobbing issue should disappear, although it may leave some (hopefully harmless) detritus next to the tower.

nsensfel, 2023-05-12:
As you mention in the last sentence of the original post, having the print head move to the side of the wipe tower prior to sending the `Tx` command to the MMU might resolve this issue, though it may also leave some (hopefully harmless) detritus laying around.
Until further options are made available in the slicer, wouldn't it be possible to make a script that replaces the `Tx` commands in the generated GCode so that this move to the side is done?

nsensfel, 2023-05-25:
If anyone is interested in trying the "raise the head off the purge block, or out to one side" solution, I have tried my hand at writing a script for that. I've actually made a little tool to help making these types of script: https://github.com/nsensfel/gcode-scripter.

The script is ran offline (i.e. not during the print), so there's no risk in trying.

Here are the instructions to run it:
1. Clone (requires Git) or download the repository.
2. Make sure you have Python3 installed (e.g. use the `python3` command in PowerShell for Windows).
3. Have the gcode-scripter folder as current directory.
4. Run `python3 ./GCodeScripter.py -e script.SaferMMYWipeTowerSwap -if PATH_TO_YOUR_INPUT_GCODE_FILE -of PATH_TO_OUTPUT_GCODE_FILE` in a terminal (e.g. PowerShell for Windows).

You should be able to drag and drop the input G-Code file to the terminal so it writes the path directly.
By default, the script will target `(X: 0.0, Y:200.0)` as the safe spot to make the swap, with an increase of `1.0` for the Z axis as it moves and swaps. This can easily be changed.

**Beware:** this is both an experimental tool and an experimental script for it. Use the G-Code viewer to check it does what you expect it to (available as part of PrusaSlicer).

If you wish to change the script's target location or Z offset, use the `-sa` parameters. It'll also check for the printer's print area, which can be changed through the `-vpa` arguments. Here is an example setting them all:
`python3 ./GCodeScripter.py -e script.SaferMMYWipeTowerSwap -if my_little_gcode.gcode -of my_altered_little_gcode.gcode  -sa "SMMUWTS_TARGET_X" "210" -sa "SMMUWTS_TARGET_Y" "205" -sa "SMMUWTS_Z_STEP" "0.5" -vpa "MAX_X" "220" -vpa "MAX_Y" "215" -vpa "MAX_Z" "230"
`

I'm still experimenting with that solution. It'll still make blobs when at the target location, and if these stay glued to the nozzle, then it's not really resolving the issue in a reliable manner.

closed 2026-09-29 by prusaslicer-bot[bot]