prusa3d/PrusaSlicer #3718: Need option to use previous file naming scheme

leftspin2k, 2020-02-25:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer-2.2.0-beta+win64-202002211801

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

The old file naming from Slic3rPE-1.42.0-beta+win64-full-201903171406 and before was great. 

1) The filename automatically tracked the current model when all models were deleted, and a new model was added. It also switched to the current model's directory (I think there is a preference for this behavior in PrusaSlicer-2.2.0-beta+win64-202002211801. 

2) Now the print time is added to the filename. This is a nice thing when you are exporting the gcode one time. When multiple iterations are made, this is really frustrating since the user must hunt down the previous versions manually to delete them. This is because the previous version may no longer match the old filename due to the change in print time.

For my designer's workflow, I iterate during the model creation: saving it out as the same STL filename, making a gcode file, print testing part of the model, test for fit, and repeat.

Please add a preference to use the old behavior, or revert to the old behavior.

_Is this a new feature request?_
yes? I want to be able to switch to the old file naming behavior from 
Slic3rPE-1.42.0-beta+win64-full-201903171406

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

JohnOCFII, 2020-02-25:
@leftspin2k - you can change the output file format to match the old way in the Print Settings dialog.  Like you, I delete the time portion (print_time).

[image]

bubnikv, 2020-02-27:
Is your problem solved? If yes, then please close this issue.

closed 2020-02-27 by leftspin2k