prusa3d/PrusaSlicer #3719: Mac column widths for Name/visible/editing are wrong

triantos, 2020-02-25:
### Version
2.1.1

### Operating system type + version
macos 10.15.3 catalina

### 3D printer brand / version + firmware version (if known)
Stock MK3S+MMU2S with latest firmware

### Behavior
When I launch PrusaSlicer or open a .3mf file, the columns on the right side of the UI for the object name / visibility toggle / editing are all strange widths. Name and visibility are both minimal side, while Editing takes up all of the remaining width. By default they should probably use minimal width for visibility and editing, just enough to show the icons, and allocate the rest of the space for the name. See attached pics. This happens both in multimaterial mode and single mode.

This is what I see on any 3mf file
[image]

This is probably what should be the default.
[image]

#### Project File (.3MF) where problem occurs
[sample.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4251985/sample.3mf.zip)

bubnikv, 2020-02-26:
Do you see the issue with 2.2.0-beta? I remember @YuSanka solved that issue once, but I don't remember in which version and she is on vacation now.

triantos, 2020-02-26:
No the beta seems correct. Sometimes this kind of configuration is stored in a plist in Mac apps but I couldn't find anything like that. The .ini file contains different information.

bubnikv, 2020-02-27:
Thanks, so we can close this issue. We plan to release 2.2.0-rc soon.

closed 2020-02-27 by bubnikv