prusa3d/PrusaSlicer #3720: Filament presets not present

nbennett1978, 2020-02-26:
### Version
2.2.0-beta+win64

### Operating system type + version
Windows 10 64Bit

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S MMU2S
### Behavior
* i installed the built in printer presets
* i then installed all built in filament presets
* later, in the slicer, none of the filament presets are shown. List is empty 
* i tried the same with MK2S and custom printer, still no filament preset available in the menu

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-02-26:
Please read the following discussion
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-2-0-beta-2/
whether it explains your situation.

nbennett1978, 2020-02-26:
yep, that was it, thanks!!!!!!

closed 2020-02-26 by nbennett1978

bubnikv, 2020-02-26:
@nbennett1978 

Do you have any opinion as of how to improve the current situation? Are you happy with the manual resolution of your problem?

nbennett1978, 2020-02-26:
Reaading the thread you sent me, they said something about a red flag, which when clicked, made all incompatible filament combinations show. I found that inconvenient, as nothing would show at all in the first place, and the flag didn’t really draw my attention. Thus I find the given situation inconvenient.

I see the problematic that filament settings, change with different nozzle diameters. I have only used 0.4 mm nozzles so far, so I have not any experience with that. It would be nice to find a way to include nozzle-dependent differences in a single filament profile for all nozzles.

Von: Vojtěch Bubník
Gesendet: Wednesday, February 26, 2020 13:26
An: prusa3d/PrusaSlicer
Cc: Nureddin Bennett; Mention
Betreff: Re: [prusa3d/PrusaSlicer] Filament presets not present (#3720)

@nbennett1978
Do you have any opinion as of how to improve the current situation? Are you happy with the manual resolution of your problem?