prusa3d/PrusaSlicer #3721: Multiple extruder temperature change on inactive tool head

Highlanderes, 2020-02-26:
### Version
PrusaSlicer Version 2.2.0-beta+

### Operating system type + version
MacOS 10.12.6

### Behavior
When slicing a dual extruder (hotend) dual material print, the inactive toolhead temperature is raised, when changing from layer 1 to layer 2, but while T1 is being active, leading to a failed print (attached example gcode line 906). When using tool change G-code as a workaround, this unexpected behaviour persists.

The T0 material has a dedicated layer 1 temperature defined, activating this behaviour.

To reproduce (gcode file included), open the attached projekt, slice and save. In gcode line 906 during T1 active, T0 temperature is suddenly raised while inactive.

The general expectation is for the slicer to refrain from changing inactive tool head temperatures outside tool changes. Also when tool change g-code is applied.

#### Project File (.3MF) where problem occurs
[220Beta_TAZ6 FDv2 Factory Printer test.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4255763/220Beta_TAZ6.FDv2.Factory.Printer.test.gcode.zip)
[TAZ6 FDv2 Factory Printer test2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4255764/TAZ6.FDv2.Factory.Printer.test2.3mf.zip)

Highlanderes, 2020-02-27:
A work-around with After Layer Change g-code unfortunately runs before the offending temperature change by the slicer. 

The other thing I can think of now (avoiding hardcoding in the startscript), is to set the same temperature for all layers in the slicer, and hardcode a higher layer1 temperature in a tool change script (to avoid touching the start script):

`M104 S175 T[current_extruder] `

`{if next_extruder == 0}M109 R{(layer_num<1 and next_extruder == 0 ? 270 : temperature[next_extruder])} T[next_extruder] {endif} `

`{if next_extruder == 1}M109 R{(layer_num<1 and next_extruder == 1 ? 245 : temperature[next_extruder])} T[next_extruder] {endif} 
`

bound to give problems later, so other suggestions are more than welcome.

lstankiewicz, 2020-05-22:
Hello,
I just finished enabling dual extruder on my printer and encountered same issue in 2.2.0 (latest release).
Same situation, different temperature for first layer and when the layer is changed Slicer cools down first extruder and at the same time heats up second one to the same temperature ignoring ooze prevention temp offset.
piece of offending gcode:

G1 X192.421 Y86.844 E279.32228
G1 Z0.400 F7200.000
M104 S243 T0 ; set temperature
M104 S213 T1 ; set temperature
G1 E277.82228 F2400.00000
G92 E0
G1 Z0.600 F7200.000
G1 X200.169 Y83.341
G1 Z0.400
G1 E1.50000 F2400.00000
G1 F4500.000
G1 X200.346 Y84.617 E1.54040

to fix that I need to edit each dual color gcode.

unixunion, 2024-09-13:
Unsure if related, but in my multi tool setup, I use G10 to configure my active and standby temps, but then PrusaSlicer overwrites all temperatures when a filemant temperature change is required for layer2. 

```
; set the initial tool up
G10 P0 S{first_layer_temperature[0]}      ; set T0 extruder temp
G10 P0 R{temperature[0] - 70}; set T0 idle temp
G10 P1 S{first_layer_temperature[1]}      ; set T1 extruder temp
G10 P1 R{temperature[1] - 70}; set T1 idle temp
G10 P2 S{first_layer_temperature[2]}      ; set T2 extruder temp
G10 P2 R{temperature[2] - 70}; set T2 idle temp
G10 P3 S{first_layer_temperature[3]}      ; set T3 extruder temp
G10 P3 R{temperature[3] - 70}; set T3 idle temp
```

But at layer2, I get this in gcode. 
```
G1 Z.52 F21000
M104 S210 T0 ; set temperature
M104 S210 T1 ; set temperature
M104 S210 T2 ; set temperature
```

There should me a option to not use M104, but use G10 instead.

closed 2026-09-29 by prusaslicer-bot[bot]