prusa3d/PrusaSlicer #3722: [Feature Request]: Add ability to change "plus" default g-code or function

McPcholkin, 2020-02-26:
### Version
PrusaSlicer-2.2.0-beta+win64-202002211801

### Behavior
I start learning multi-color printing and found that "plus"  button is very limited to use.
In new 2.2.0 version it much better but not comfort to use.

#### Problem:
Octoprint need M0 command to pause print (color change, magnet insert, etc) but slicer can add only M600
So for now i have 2 solutions:
1 - Right click on "plus" button => "Add custom G-code" => type on keyboard "M0" => OK button
2 - Write script that find and replace M600 command to any i need.
[image]

I Write [script ](https://github.com/McPcholkin/PrusaSlicer-CreawsomeMod/blob/master/post-processing/PrusaSlicer_M600_to_M0.py)but much better be implement ability to change default "plus" button function in interface.

### I can propose couple variants how it may be implemented:
Move color change G-code variable to "Custom G-code" in "Printer settings" tab.
[image]

OR

In slicer "Preferences" menu add "Plus button function" dropdown menu.
[image]

OR

Add "Color change G-Code:" dropdown menu in "Printer settings" tab.
[image]

Any of this changes fix this problem and give flexability to slicer configuration.

neophyl, 2020-02-26:
What about those users who use all 3 options ?  Are we supposed to change preferences, use the + button, then change the preferences again and then click the button again and so on ?

How are we supposed to insert a colour change at layer 20, then add some custom gcode at layer 40 and then add another colour change at layer 100 ?

bubnikv, 2020-02-26:
I think it is sensible to add the color change and pause G-code templates to printer configuration.

McPcholkin, 2020-02-26:
> How are we supposed to insert a colour change at layer 20, then add some custom gcode at layer 40 and then add another colour change at layer 100 ?

For example  pres right mouse button on "plus" in menu select between variants:
* insert pause
* insert color chandge
* insert custom G-code

And "plus" button remember last used insert, to indicate what chosen now maybe need change color or shape of point on ribbon.

I just share my suggestions and ideas, feel free to share yours.

neophyl, 2020-02-26:
Thats cool, from your original description thats not how I'd read it.

McPcholkin, 2020-02-27:
> As well as the colour change, pause and custom, add in custom 2 and custom 3 too.

Like this?

> Maybe something lite this:
> [image]
> 
> Work like previous but add 3 template inserts, it can be added to "Custom G-code" in "Printer settings" tab.
> [image]
> If nothing put in template field it can be excluded from menu.

neophyl, 2020-02-27:
Yeah that's pretty much what I was thinking.

sej7278, 2020-05-02:
What about with sailfish firmware? There you seem to have to calculate the height in mm (layer height * layer number) and call `M322 Zzzz.zz`

https://www.sailfishfirmware.com/doc/details-special-gcode-commands.html

So as its not a simple drop-in-replacement for M600 it couldn't be put into the profile as its needs a function to calculate it.

bubnikv, 2020-10-09:
OIriginal request implemented with 43cfd4486485ebc11dec4d01f321f8d760e82aa2

The upcoming PrusaSicer 2.3.0-alpha1 will support custom G-codes for
"color_change_gcode" "pause_print_gcode" "template_custom_gcode"

Closing.

closed 2020-10-09 by bubnikv