prusa3d/PrusaSlicer #3725: [FR] Psuedorandom seam position

fiveangle, 2020-02-26:
### Version
2.10 win64 201909160915

MK3 3.18

This is a new feature request.

### Problematic Behavior
* Some concentric-dominated prints require dimensional accuracy.  Currently, the only viable "Seam position" for parts that require exacting dimensional accuracy is "Random", however the truly random seem generation results in variable Z-blob and unnecessary lengthening of print time due to the truly random moves between Z layer changes. The truly random nature of the "Random" seam position is also technically not as dimensionally accurate if the print consists of a relatively small number of layers, since only with large numbers of layers does the "Random" seam positions randomize out to unity around the print perimeter.

One very real example I just encountered is printing concentric rings for centering automobile wheels on hubs that are of a difference size.  Printing anything but "Random" would result in a scar that prevents the part from being installed, while the "Random" option resulted in various different Z-blobs that at least would work, but required light sanding to knock down only the largest Z-blobs:

[image]

### Suggested Resolution
* As an alternative to "Random" I suggest a "Psuedorandom" option that moves a fixed but relatively short distance along the part and then begin the next layer.  This will result in Z-blob minimization since the slicer retraction/coast/etc settings for the "Psuidorandom" seem postion option can be tuned to the fixed offset length.  This also minimizes the time the nozzle will be under retraction condition so that pressure remains consistent compared to "Random".  Z-blobs will also be distributed equally around the part perimeter (some logic should be included to ensure the offset does not re-align upon itself for extremely small parts and if so, simply reduce the fixed offset by 1/2 or some other chosen fixed offset (whatever is empirically determined to be a good choice).  Alternately, a user preference for the offset could be created, although I question if this is necessary since I don't see a strong condition where user manipulation of the seem to exacting psuedorandom would be necessary.  For prints that consist of a very small number of layers, one may think it necessary to move the postion to locations equally spaced around the perimeter, but I challenge that requirement since the number of layer changes is very small so should not alter the dimensional accuracy much if at all.

Thanks !

-=dave

BeatSlayer, 2020-03-03:
From my experience the nearest position advances a segment each layer when printing cylindrical/round shapes making something similar to your proposal.

lbussy, 2020-03-03:
Can get behind this.  I've noticed the same issues.

fiveangle, 2020-03-09:
> From my experience the nearest position advances a segment each layer when printing cylindrical/round shapes making something similar to your proposal.

This setting most often results in stacking of the latter change in such close proximity as to be only slightly better than a fixed seam position.

closed 2026-09-29 by prusaslicer-bot[bot]