prusa3d/PrusaSlicer #3726: Changing displays causes intermittent hangs (crashes)

Pun-e, 2020-02-27:
### Version
2.2 A4

### Operating system type + version
Win 10 64 Pro - Surface Pro

### Behavior
I start the slicer, and everything is fine.

If I unplug my docking hub, sometimes (it used to be everytime, but it has improved recently), I'm presented with this:
[image]

This is PrusaSlicer scaled to my Surface's screen, failing to update to the new resolution of my external monitor.

I'm not sure what information you need from me to properly diagnose this issue, but just let me know. It's a very circumstantial bug but it always pops up when I'm least expecting it.

bubnikv, 2020-02-27:
Do you have your surface display active in addition to your external
display? Or is just the external display active when plugged into the
docking station?

Pun-e, 2020-02-27:
Undocked = Surface in-built screen
Docked = 2 external HDMI monitors

bubnikv, 2020-02-27:
Thanks, we have a windows laptop with a docking station with 2 display
outputs available, we can test it.

What is the model of your Surface? Is it a tablet style computer or a
laptop?

Pun-e, 2020-02-27:
[This model](https://en.wikipedia.org/wiki/Surface_Pro_4) i7 + iris 540 graphics
[image]

The next time it happens, is there any log or something I can get that will offer some further details / insights?

bubnikv, 2020-02-27:
We were able to reproduce the issue on one setup of one of our IT guys. We
were not able to reproduce the issue on our development laptop. It will
take some time to resolve the issue as we cannot reproduce on our
development computers.

bubnikv, 2020-02-27:
It happened when our IT guy put his computer to sleep after running slicer
on his laptop in standalone mode, and then he plugged his computer into his
docking station and then he woke up the computer by the power on button on
his docking station. Is this your scenario as well?

Pun-e, 2020-02-27:
I'm not too sure what "standalone mode" is, but yeah, going to sleep seems to make it more likely to happen. Although I can cause the issue to happen just by toggling power to my dock. When power is switched off, it counts as a disconnect. I literally just froze the slicer again testing that.

It also seems more likely to happen after I slice something.

Yeah, just tested that too. If I just open the program and don't slice anything it didn't freeze after 3 toggles, sliced a file and it froze on the next connection.

Process:

1. **undock** device
2. Open slicer
3. Insert this file [Github wouldn't let me attach as too big, 60MB+)
4. Slice
5. Connect dock

For me, this will freeze it every time. Sometimes I can use slicer for a few seconds before it freezes but I've successfully frozen it 4 times in a row now

bubnikv, 2020-03-10:
We reproduced it on one laptop. When we tried to collect crash logs using

https://docs.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps

and to trigger the crash dump manually over the hanging process with

https://docs.microsoft.com/en-us/sysinternals/downloads/procdump

Create the process dump by executing the following command, or the one provided by a helper:
 
ProcDump.exe -ma <process name or PID (Process ID)>

we are no more able to reproduce the hang once we enabled crash dump collection for PrusaSlicer, so we are stuck. You may try to collect the crash dump for us if you manage.

closed 2026-09-29 by prusaslicer-bot[bot]