prusa3d/PrusaSlicer #3727: Consistent crashing when creating multiple height range modifiers

mooluk26, 2020-02-27:
### Version

2.1.1+

### Operating system type + version

macOS Mojave - Version 10.14.6

### Behavior

Software consistently crashes without warning when attempting to add further height range modifiers to an object before allowing the preview to 'update' with the previous modifier (NOTE: this doesn't require the preview to be in slice view, or for the modifier to actually do anything that would affect the model). This can be reproduced consistently with the following steps:

1. Select object in sidebar
2. Right Click > 'Height range Modifier'
3. Enter values for 'Start at height', 'Stop at height' & 'Layer height' (0.15 / 0.25 / 0.1 in example file)
4. Click anywhere in 3D view to confirm these changes. **NOTE:** This does NOT cause a crash.
5. Select 'Layers' row in sidebar below previously selected object
6. Click the '+' in 'Height ranges' selection below to add a further range.
7. Enter values for 'Start at height', 'Stop at height' & 'Layer height' (0.25 / 0.35 / 0.1 in example file) DO NOT click anywhere else other than these fields or use the tab key
8. After completing these fields, click the '+' to add a further height range. This will instead cause a crash 100% of the time, displaying the macOS 'PrusaSlicer quit unexpectedly' screen immediately.

Please note that the crashes only occur if you **DO NOT** click back in the main preview area after adding each height range.

#### Project File (.3MF) where problem occurs

[PickExample.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4261425/PickExample.3mf.zip)

bubnikv, 2020-02-27:
Please test with 2.2.0-beta. Thank you.

mooluk26, 2020-02-27:
I can confirm that this also happens with 2.2.0-beta having just tried it.

Many thanks

bubnikv, 2020-02-27:
This is a weird crash that happens somewhere inside wxWidgets / Cocoa likely due to some handle that is no more valid.

[image]

bubnikv, 2020-02-27:
These kind of issues are a developer nightmare, it will take quite some time to track down the culprit.

bubnikv, 2020-03-11:
This will be fixed in PrusaSlicer 2.2.0-rc3.
Closing.

closed 2020-03-11 by bubnikv