prusa3d/PrusaSlicer #3728: Overlapping Nearly Redundant SLA Support Trees

NamedJason, 2020-02-27:
### Version
2.1.0

### Operating system type + version
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
When generating support trees for complex objects (this is an archer miniature from https://www.thingiverse.com/thing:3892394), the generated support trees overlap in situations where I would expect them to just use one existing tree (see all of the overlapping vertical trees in this screenshot).
[image]

This abundance of trees uses more resin while printing and increases the difficulty when removing supports from the printed objects.

Is it possible to change the tree generation behavior to encourage the consolidation of trees or the re-use of existing trees in order to minimize the excessively generated tree structure?

#### Project File (.3MF) where problem occurs
[Bandit_Archer_1.zip](https://github.com/prusa3d/PrusaSlicer/files/4263135/Bandit_Archer_1.zip)

tamasmeszaros, 2020-03-02:
Hi,
I've added a parameter called "Max bridges on a pillar" with which you can influence how much pinheads can be routed to existing pillars with a bridge. This could reduce the amount of support structure material.

Also you can try to play around with the "Max bridge length" and "Max pillar linking distance" parameter, to influence the length of these bridges and cross-links between pillars.

Reducing the amount of support points also helps. This can be achieved with the "support point density" and "Minimal distance of support points" parameters.

Good luck!

closed 2020-03-02 by tamasmeszaros

NamedJason, 2020-03-02:
Thanks, I'll give those settings a try!

commit 090ce6ca (2020-03-02): Promote max_bridges_on_pillar to be a runtime parameter.

This way the user greater control over support tree branching and the amount of pillars created.

fixes #3728