prusa3d/PrusaSlicer #3730: Toolpath outside the print area in Z with with 210mm height 3d models

belgiancoder, 2020-02-27:
### Version
2.1.1 (slicing for MK3 with 0.15 layer height)

### Behavior
* Toolpath outside the print area (in the Z axis) when the model fits in Z
* _Steps needed to reproduce the problem_
  * Make a model that is 21cm heigh
  * Slice it with 0.2 first layer height, 0.15 layer height
* _Expected Results_: It slices
* _Actual Results_: It doesn't slice (Toolpath outside the print area)
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

It doesn't slice because although the model is 21cm heigh and the layer height is .15, since the first layer height is 0.2, it makes 1399 layer at 0.15 and 1 layer at 0.2, it total, that is 21.05cm.

[image]
[image]

So I thought OK, I will make my layer height 0.1499642602, and guess what ! It WORKED !

[image]
[image]

bubnikv, 2020-02-28:
How about the Z-hop to move away from the print at the end of the print? I
suppose the Z-hop will not fit into your Z space.

belgiancoder, 2020-03-02:
It just won't do the hop, the hop is conditional and only done if possible.

bubnikv, 2020-03-03:
> It just won't do the hop, the hop is conditional and only done if possible.

I am not talking about the final lift of the print head, but about the Z-hop between extrusions on the same layer.

belgiancoder, 2020-03-03:
Great question, the only thing I can tell you is that it printed ok.

belgiancoder, 2020-03-14:
@bubnikv Found why !
[image]

bubnikv, 2020-03-15:
so is your problem solved?

belgiancoder, 2020-03-15:
No, I only answered your question from 12 days ago.
The problem in the first post was that the last layer of a 21cm object has 50% chance of being outside the print area.

bubnikv, 2020-03-15:
so it will work for you if you increase the build volume maximum Z to 21cm plus 0.1mm?

belgiancoder, 2020-03-15:
Yes, it slices ! (Haven't tried printing this though)

belgiancoder, 2020-03-15:
Maybe it should compute the last layer height to arrive at the right total height ?
Maybe it's not a real issue and should be closed, I don't know.
But it took me a while to find what the issue was.

bubnikv, 2020-03-15:
this is an everyday's nemesis of somebody working with floating point arithmetics.
We may add an epsilon to the max print height to avoid these issues.

closed 2026-09-29 by prusaslicer-bot[bot]