prusa3d/PrusaSlicer #3733: Viewport rotation intermittent freezing

WarrenSchultz, 2020-02-28:
### Version
2.2.0-beta+win64

### Operating system type + version
PrusaSlicer
Version:   2.2.0-beta+win64
Build:     PrusaSlicer-2.2.0-beta+win64-202002211801

PrusaSlicer
Version:   2.2.0-beta+win64
Build:     PrusaSlicer-2.2.0-beta+win64-202002211801

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition
Total RAM size [MB]: 68,369MB
OpenGL installation
GL version:   4.6.0 NVIDIA 442.19
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
[... 336 lines cut ...]
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
All printer presets

### Behavior
When rotating the viewport it will randomly freeze the PS viewport as well as the rest of the app. 
This did not occur on the 2.1.x release versions.
Simple to reproduce (on my machine, anyway). Left-click and drag back and forth in the viewport to rotate the empty print surface around. Eventually, the view will freeze for five or more seconds, and then eventually resume. 
I saw this on earlier versions of the 2.2 alphas, but thought I saw an issue already filed, however I couldn't find one today about it.
I was on a previous display driver revision, but updated to the latest nvidia studio release, and the issue persists.

bubnikv, 2020-02-28:
Do you have a 3DConnexion (the 3D mouse) connected? Or do you use just the
2D mouse?

bubnikv, 2020-02-28:
Do you have the free or constrained mouse rotation enabled in the
preferences?

bubnikv, 2020-02-28:
Your issue sounds related to https://github.com/prusa3d/PrusaSlicer/issues/3515

bubnikv, 2020-02-28:
Is your mouse cursor moving smoothly when the viewport rotation hangs? Please see my response at #3515

WarrenSchultz, 2020-02-28:
It could be related to the polling, but not the same issue as far as I can tell.
No 3D mouse connected.
I am not using perspective or free camera.

bubnikv, 2020-02-29:
>  No 3D mouse connected.

PrusaSlicer polls for the 3D mouse anyways.

WarrenSchultz, 2020-02-29:
Sorry, I edited after the original post, so I don’t think you got the full message.

bubnikv, 2020-03-01:
We are working on a background thread solution, we will likely have a prototype on Monday or Tuesday. We would ask you to test it then.

bubnikv, 2020-03-07:
We have reworked two services that were introduced in PrusaSlicer 2.2.0-alpha: Detection of 3DConnexion USB plug / unplug events and removable USB driver or SD card plug / unplug events. These detection tasks were performed at a user interface thread, while now they are executed in their respective background threads in PrusaSlicer 2.2.0-rc2. Plase let us know whether this refactoring resolved your issues. We are closing this issue. Please feel free to reopen if your issue has not been resolved with PrusaSlicer 2.2.0-rc2.

closed 2020-03-07 by bubnikv

WarrenSchultz, 2020-03-07:
This appears to have resolved the issue so far, thank you!

bubnikv, 2020-03-08:
thanks for heads up. I am quite relieved as debugging such intermittent issue at customer's computer is very difficult.