prusa3d/PrusaSlicer #3736: Moving holes after previewing hollowing does not work correctly

ardenpm, 2020-02-29:
### Version
2.2.0-beta+win64

### Operating system type + version
Windows 10 (1903)

### Behavior
* When moving holes on hollowed geometry, if the hollowing preview has been created the new hole snaps to the hollowed preview geometry instead of the original geometry. Here is the normal behaviour before previewing:

[image]

And after preview geometry is generated:

[image]

Notice how the hole _falls into_ the original hole when moving the hole around. This makes it very difficult to make slight adjustments to the hole position if you've previewed the hollowing.

* To reproduce add some geometry (just a cube is fine), set it up for hollowing and place a hole but don't enable hollowing preview yet. Drag the hole around, observe it moves smoothly. Now turn on the hollowing preview, observe the hole position becomes eratic when the cursor goes over the existing hole location.

* When dragging a hole it would be expected that the existing hole would not interfere with the movement.
* Actual result is that the existing hole does interfere with the movement.

lukasmatena, 2020-03-18:
I understand the problem but the solution is more complicated than it looks. The reason the hole sticks to the updated geometry is that it must be possible to place the hole on the interior of the cavity. And because the updated geometry also contains the hole, the result is what you see.

What you are proposing would mean to keep the inside and outside walls separate and track the mouse along both meshes separately. It will also probably mean to make another copy of some mesh internally (because of the clipping plane, which needs the meshes combined). I'm not sure if it's worth the effort. Currently the fine adjustment of the holes must be done before the preview (or after it is invalidated again). If we decide to change it, it will not be in 2.2.0 release.

ardenpm, 2020-03-18:
Can see the complexity there. One thing that would help I think when moving holes in the invalidated state at least would be to have an outline or ghost of where the hole was before.

Often when I am moving a hole, I want to do so relative to the existing placement, so it helps to see where the hole was. With that addition I think invalidating the hollowing to move the holes would work well.

So I could then uncheck/check the _Hollow this object_ box to invalidate and drag holes around but still know where they were before I started dragging. That would be a good compromise I think.

closed 2026-09-29 by prusaslicer-bot[bot]