prusa3d/PrusaSlicer #3737: Import 3mf file from QIDI Printer 5.3.2

RicardoGonda, 2020-02-29:
### Version
_Version of PrusaSlicer used goes here_
2.1.1+Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

I just download and Install the PrusaSilicer 2.1.1+Win64
My previews  printer is QIDI X-PRO
I have a lot of 3MF files that I want to start printing with my new Prusa 
I try to open several of them, and receive the same error 
I need HELP !!!

Error revived wile try to import a 3mf file generated by the QIDI slicer 
**PursaSlicer error Error (parsing aborted) while parsing '3D/3D model.model' at line 9**

[image]

bubnikv, 2020-02-29:
Please attach couple of your 3MFs, otherwise we cannot do anything.

RicardoGonda, 2020-02-29:
Dear Vojtěch 
Thank you for your answer
I just find the problem. QIDI Slicer hava a option to GROUP the several parts of the project. I just save the project UNGROUP and PrusaSlicer read the 3MF file without a problem. 

[image]

bubnikv, 2020-02-29:
ok. Please test whether your 3MF files are loadable by Cura and Microsoft
3D Builder. Those two applications are a benchmark to us.

rtyr, 2020-05-21:
I was able to reproduce it with Qidi Print slicer using group option. I believe that exported 3MF is not in correct format, because it is impossible to open it in any software, except Qidi and Cura (I tried Simplify3D, Netfabb, Microsoft 3DBuilder, Autodesk Meshmixer). 

Qidi slicer looks like Cura with different skin, so it is not surprising that Cura can open that file. With grouped objects on the plate, Cura exports similar (invalid) 3MF file. Closing.

closed 2020-05-21 by rtyr