prusa3d/PrusaSlicer #3741: Support enforcers should ignore "don't support bridges"

wavexx, 2020-02-29:
### Version
master @ c9a75bb70a2c1c88e72e0dd55961b5701431f0a6

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
If I use a "support enforcer" under a long bridged section there's explicit intention from the user to create a support area in the request volume.

Currently you have to disable "Don't support bridges" in order to be able to do so, which will create supports on all other bridged sections, requiring you to use support blockers everywhere else.

A support enforcer should ignore "don't support bridges".

Related issues where this is also mentioned:

https://github.com/prusa3d/PrusaSlicer/issues/2152
https://github.com/prusa3d/PrusaSlicer/issues/1326

mlazzarotto, 2020-06-30:
Can confirm, I'm having the same issue with PrusaSlicer 2.2.0.
Support enforce seems to have zero effect on support generation.

n8bot, 2020-06-30:
I'm no expert on using enforcers and blockers, or even modifiers... but could you not use a modifier instead of an enforcer here? If the option to change the "Don't support bridges" with the modifier exists, this should be no problem and then supports will automatically generate there (if it all works as I assume it does).

wavexx, 2020-06-30:
I don't see it as an available modifier, but even then you'd still need both an enforcer and a modifier on the same location to get the desired result.

mlazzarotto, 2020-06-30:
@n8bot that is a good suggestion, but I can't change "Don't support bridges" on the modifier.

n8bot, 2020-06-30:
That's too bad. I think the original suggestion to have enforcers override this option may be a good idea. At the same time, the option to modify support settings with modifiers would be nice too.