prusa3d/PrusaSlicer #3742: modifier box for adding "Detect Thin walls" causes printing the modifier box itself

3Dietrich, 2020-02-29:
### Version
2.2.1+
### Operating system type + version
MacOS Sierra 10.12.6
### Behavior
#### _Describe the problem_
I wanted to add a modifier to do "Detect thin walls" at a certain position only. The actually wanted walls are fine but there are additional walls (double thick) of the modifier box too.
#### _Steps needed to reproduce the problem_
_short:_
load the attached .3mf and switch the "Detect thin walls"  in the modifier off and on

_verbose:_
- an object without Infill (and top is open like a Vase but with more freedom of perimeters)
- add a "Generic Box" modifier on an object
- add Add Settings/Layers and Perimeters/ --> check "Detect thin walls"
- switch "detect thin walls" in the option to active
#### _Expected Results_
Thin walls add/subtract the opention of single layerwidth walls, nothing else
#### _Actual Results_
The thin walls are created but the edge of modifier box is part of print too (and regarding the wanted function: no 'thin' but 2 times of wall thickness! :)
_Screenshot: normal, without 'thin walls'_
[image]
_Screenshot showing the bug (the added 4walls are not wanted)
[image]
_Is this a new feature request?_
no, it's a bug I guess
#### Project File (.3MF) where problem occurs
[added_awhole_DetectThinWalls-Modifier-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4270271/added_awhole_DetectThinWalls-Modifier-Box.3mf.zip)

bubnikv, 2020-03-17:
This issue is similar to
https://github.com/prusa3d/PrusaSlicer/issues/3846

The way the modifier volumes work is they split the object into pieces, which are sliced independently. Thus the extra perimeter lines added.

The way you expect PrusaSlicer to override number of perimeters is significantly more involved to implement than what we currently have.

3Dietrich, 2020-03-17:
call my petty but I don't expect "to override number of perimenters" but "ignore the modulators outside surface". There are just no perimeters to be overridden, the modulator is for changing  a complete different setting, not adding any surface or perimeters..
Hoever, the rest ("is significantly more involved to implement than what we currently have") I need to accept.

A workaround can be to make the modifier bigger than the object, this works for the attached example at least.

bubnikv, 2021-06-03:
Duplicate of #3370

closed 2021-06-03 by bubnikv