prusa3d/PrusaSlicer #3743: libgmp-10.dll and libmpfr-4.dll missing when compiling PrusaSlicer

martijnschouten, 2020-02-29:
### Version
Using master branch at 21:31 at 29-02-2020 (version 2.2.0 RC)

### Operating system type + version
I'm running and up to date version of windows 10 pro.

### Behavior
I have compiled the dependencies and run cmake for building PrusaSlicer, as described [here](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md):

However I did get a "PrusaSlicer.dll was not loaded" error, similar to the issue [2584](https://github.com/prusa3d/PrusaSlicer/issues/2584). After inspection with  Dependecy Walker a lot of files are missing. The usual suspects such as API-MS-xxx.dll,  EXT-MS-xxx.dll, EMCLIENT.dll, hvsifiletrust.dll and IEShims.dll. But also libgmp-10.dll and libmpfr-4.dll. I had to copy these last two files from the 2.2.0-RC relase zip and place them next to my  build/src/debug folder for the PrusaSlicer debug configuration to to run.

tamasmeszaros, 2020-03-20:
Hi @martijnschouten ! Did you try to do a cmake configure and build from scratch by deleting cmake cache?

martijnschouten, 2020-03-20:
I am not sure that I understand what you are asking. My logbook of that day says I did the following (which is pretty much what is in the noob guide in the installation instructions).

mkdir build

cd build

cmake .. -G "Visual Studio 16 2019" -DDESTDIR="c:\PrusaSlicer\PrusaSlicer-master"

msbuild /m ALL_BUILD.vcxproj 

cd c:\PrusaSlicer\PrusaSlicer-master\

mkdir build

cd build

cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="c:\PrusaSlicer\PrusaSlicer-master\usr\local"

install visual c++ for developers by running the visual studio installer again (was configured for c#)

Open->Project/Solution or File->Open->Project/Solution

click on c:\PrusaSlicer\PrusaSlicer-master\build\PrusaSlicer.sln

copy libgmp-10.dll and libmpfr-4.dll from a standard build zip

build by clicking on the debug button in visual studio

tamasmeszaros, 2020-05-18:
Do you have `SLIC3R_STATIC=1` in your environment? If not, set it or call cmake configure step for the project build (not the deps build) with an additional `-DSLIC3R_STATIC=ON` parameter.

ardenpm, 2020-07-19:
I have the identical issue and have also tried with `-DSLIC3R_STATIC=ON` but this did not make a difference for me (if the static option is required shouldn't it be on by default or in the build instructions).

dirrgang, 2020-08-08:
I seem to have the same issue. The program fails to start with "PrusaSlicer.dll was not loaded" after building it.

Richie76, 2020-10-14:
Same issue here - followed each step - also tried the static option.

Can anyone here give some advise ?

bubnikv, 2020-10-15:
We took them from the Windows CGAL installer.
Just copy the files manually from our PrusaSlicer binary distro.

Richie76, 2020-10-15:
I managed it to get it running without copying the dll.

I followed the noob instructions.

after loading the SLN-file into visual studio 19 i changed the "version" from debug to release and and used Build > Build solution and let this run.

Then i went to the folder "D:\PrusaSlicer\PrusaSlicer-master\build\src\Release" (drive letter and folders might be different) ... and there are many files including the dll´s and the prusaslicer.exe which i started and used.

Hope this works for you as well.

foreachthing, 2020-10-15:
To get me startet on different machines, I wrote this small batch (_basically the "noob instructions"_) file (assumed you're using VS2019 x64):

Make sure you start this batch in `Developer PowerShell for VS 2019`!

<pre>@echo off

ECHO I'm going to CLONE the PrusaSlicer repository - ready?
ECHO Best, if you're curently in c:\dev (or similar short path)
ECHO Also make sure you got at least 15 GB of free disk space available
ECHO.
ECHO Press CTRL+BREAK if you want to cancel.
PAUSE
ECHO Are you sure??? Shall I continue?
PAUSE

git clone https://github.com/prusa3d/PrusaSlicer.git --recursive

cd PrusaSlicer

SET "thisPATH=%CD%"
echo Building in %thisPATH%

cd deps
mkdir build
cd build

cmake .. -G "Visual Studio 16 2019" -DDESTDIR="%thisPATH%"

REM This will take a loooong time! Go get some coffee...
msbuild /m ALL_BUILD.vcxproj

cd "%thisPATH%"
mkdir build
cd build
cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="%thisPATH%\usr\local" -Wno-dev -Wno-error=deprecated

msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal
</pre>

Now you could add this as well:
`robocopy "c:\dev\PrusaSlicer\build\src\Release" "c:\PrusaSlicer_Alpha" /MIR /xf *.exp *.lib *.pdb /R:600 /W:30`

Use at your onw risk ;-)

rackley096794, 2020-11-03:
I created a PR that should address this issue (added Build->Rebuild Solution among many other things).

bubnikv, 2021-08-13:
Merged, closing. Thanks for your contribution.

closed 2021-08-13 by bubnikv