prusa3d/PrusaSlicer #3744: 2.2.0-rc Ensure vertical shell thickness setting not honored for single wall extrusions.

skellatore, 2020-03-01:
### Version
2.2.0-rc+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s

### Behavior
Ensure Vertical Shell thickness feature  - When unticked, is not honored with single perimeter/single wall extrusions.

This leads to a lovely 2+hour increase in time in this scenario and a poorer quality print.

#### Project File (.3MF) where problem occurs
[vertical shells.zip](https://github.com/prusa3d/PrusaSlicer/files/4271289/vertical.shells.zip)
[image]
[image]

JohnnyDeer, 2020-03-02:
Hi, this is known issue and is on our to do list.

skellatore, 2020-03-03:
It actually does it even if there is more than than just single wall.. I've got this case now..

[image]

This is a really, terrible bug.  Excellent software let down by a feature that doesn't follow the rules of its own checkbox.

Also, i've found on some prints that the slicer doesn't disable the bridge flow multiplier for subsequent/same layers. And on those layers you get over-extrusion on perimeters. Setting the value to 1 (ie, dont use it) fixes this.

bubnikv, 2020-03-04:
We inherited many features and bugs from the upstream slic3r and we are working on them relentlessly. The "ensure vertical wall thickness" feature was implemented as an alternate implementation of the solid infill on sloping surfaces. When the check box is unchecked, the old algorithm from upstream Slic3r is in action.

As a workaround you may use the layer modifier to set the top and bottom layers to zero.
[image]

[vertical shells-w-modifier.zip](https://github.com/prusa3d/PrusaSlicer/files/4287331/vertical.shells-w-modifier.zip)

We plan to dive into this issue, though there is so much to do.

bubnikv, 2020-03-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/668

skellatore, 2020-03-08:
@bubnikv Thank's for providing a work-a-round. I never considered this.

bubnikv, 2020-03-08:
I am sometimes surprised myself how our content guys or our customers are
creative with the modifiers.

haleypearse, 2020-08-25:
I use this modifier trick all the time, but it is very dumb to have to place modifiers over sloping surfaces but not actual top/bottom parts of every part. I also use low layer heights for the same effect, but this leads to increased visibility of [VFAs](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../stepper-motor-upgrades-to-eliminate-vfa-s-vertical-fine-artifacts/) due to the lack of geared extruder on the prusa mk3.

robustini, 2020-08-26:
There's no solution to this problem?

xionic, 2021-05-21:
Thanks for the work so far guys. I encounter this bug all the time, it's super annoying and the workaround is not always practical. Any updates on when this will be looked at?

KevAquila, 2022-07-16:
Just ran into this.  Figured I'd say it still exists.

napter, 2022-10-05:
This is still an issue in 2.5

closed 2026-09-29 by prusaslicer-bot[bot]