prusa3d/PrusaSlicer #3745: master branch wont build

jarl-dk, 2020-03-01:
### Version
master (c9a75bb70a2c1c88e72e0dd55961b5701431f0a6)

### Operating system type + version
Ubuntu 19.10

### Behavior
the source code won't build according to instructions due to this
https://github.com/prusa3d/PrusaSlicer/commit/c9a75bb70a2c1c88e72e0dd55961b5701431f0a6#diff-dc899be8f29c958f4922c223289ff2a9L45

lukasmatena, 2020-03-03:
Did you try to clean cmake cache and do a clean build? Implicit wxString to std::string/const char* were disabled with 7c55674. I can imagine that the error that you mention might appear if they stayed enabled.

jarl-dk, 2020-03-03:
@lukasmatena : Did _you_ try a clean build? Does it compile for _you_ on a clean build?

jarl-dk, 2020-03-03:
OK. I can see that todays master has even more errors. Apparently there is no culture for keeping master in a buildable shape, sorry for disturbing.

closed 2020-03-03 by jarl-dk

lukasmatena, 2020-03-03:
> @lukasmatena : Did you try a clean build? Does it compile for you on a clean build?

Sure I did. And yes, it does compile for me, clean build or not.

> OK. I can see that todays master has even more errors. Apparently there is no culture for keeping master in a buildable shape, sorry for disturbing.

Master **is** buildable. It has been build several times on our buildserver since the commit that you mention and six developers are continuously working on it and building it. And they were for the past two days with no issues, so save your ironic remarks for someone else. I do not know what is wrong, but it is most likely on your end.

lukasmatena, 2020-03-03:
What version of wxWidgets do you link against? Because now that I think about it, I believe that unsafe wxString conversions cannot be turned off in wxWidgets 3.0. That might be it.

lukasmatena, 2020-03-04:
I'll answer myself. This was indeed caused by using wxWidgets 3.0 and it should be fixed on master by https://github.com/prusa3d/PrusaSlicer/commit/eb3b65d8afcc7026cec0b2054152d78e08673041.

jarl-dk, 2020-03-04:
OK, sorry @lukasmatena . I was a bit too fast there, sorry! I couldn't find any build-server for the project (I would have expected that as a badge in the README.md file) so I made a rong conclussion, sorry. Is the build server public? like travis or circle-ci or something else.
OK, back to investigation: I can see that you have found the answer your self.
But maybe you can tell me how can I figure out which wxWidget library that I link against? I tried to grep for wxwidget in the build-directory to find something in Make of CMake files, but couldn't find any.

jarl-dk, 2020-03-04:
Affter pulling the latest master, it build fine.

bubnikv, 2020-03-05:
There is a command line utility on Unix systems
https://manpages.debian.org/testing/wx-common/wx-config.1.en.html

lukasmatena, 2020-03-05:
Just to expand on previous answer:

>  Is the build server public? like travis or circle-ci or something else.

It is not public. Setting a public build system like Travis for several different platforms would take time that we so far use differently. We are glad that we managed to set up what we currently use, it is not simple.

As for how to build, I would recommend to first build dependencies independently of system-provided libraries and then link to them statically. This solves surprises like missing libraries, incompatible versions and so on. This process is automated, see https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md .

bubnikv, 2020-03-05:
Also our internal build servers are much faster than what the free services
could offer.

jarl-dk, 2020-03-07:
@lukasmatena  Thanks for sharing the inside information. Even more outsiders outside wannabe-developers will see it if just a few lines of that bulid-setup (with internal CI-server) was explained in a `README.md` here: https://github.com/prusa3d/PrusaSlicer/tree/master/doc
Thanks for sharing with me at least.