prusa3d/PrusaSlicer #3746: PrusaSlicer 2.2.0 very slow to respond

Eddyg87, 2020-03-01:
### Version
Version 2..2.0-rc+win64

### Operating system type + version
PrusaSlicer
Version:   2.2.0-rc+win64
Build:     PrusaSlicer-2.2.0-rc+win64-202002291006

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 17,171MB
OpenGL installation
GL version:   3.0 Mesa 12.0.0-rc2
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)
GLSL version: 1.30
Installed extensions:

GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_APPLE_packed_pixels
GL_APPLE_vertex_array_object
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_MESA_ycbcr_texture
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

Graphics Card Radeon RX580 with an older i7 920 CPU.

### Behavior

I just updated to Prusa Slicer 2.2.0-RC from 2.1.1. and the whole programme has become very very slow to respond. It takes 2-3 seconds per mouse click to respond and do anything. I can occasionally get the plater to rotate but it is very slow.

I had no issues with Prusa Slicer 2.1.1. and previous version to this update. I have cleared out the profiles folder and allowed that to reset it made no difference. I tried loading from cmd prompt with software rendering disabled it did not help. (--sw-renderer)

Not sure what else to try to see what may be causing it or what to try next but its not really usable at all with it being so slow to respond. I have managed to load a 3mf file with it and it did slice at normal speed but the whole interface is very slow to respond.

Apologies if i have missed some information its my first time reporting anything on here!

bubnikv, 2020-03-02:
This is likely a duplicate of
https://github.com/prusa3d/PrusaSlicer/issues/3733 and https://github.com/prusa3d/PrusaSlicer/issues/3515. 
We are working on a fix for 2.2.0 final.

bubnikv, 2020-03-07:
We have reworked two services that were introduced in PrusaSlicer 2.2.0-alpha: Detection of 3DConnexion USB plug / unplug events and removable USB driver or SD card plug / unplug events. These detection tasks were performed at a user interface thread, while now they are executed in their respective background threads in PrusaSlicer 2.2.0-rc2. Plase let us know whether this refactoring resolved your issues. We are closing this issue. Please feel free to reopen if your issue has not been resolved with PrusaSlicer 2.2.0-rc2.

Eddyg87, 2020-03-07:
Yes this has resolved my issue and it appears to be behaving properly.

Thanks!

bubnikv, 2020-03-07:
Great! Thanks for heads up.

Eddyg87, 2020-03-11:
After a few more days on 2.2.0-rc2 I have found clicking on file takes a few seconds to respond with the menu drop down. Edit, Window, View, Config, Help does not display this delay. 

Oddly i get the same behaviour when go to select the extruder for multiple bodies in one part for MMU prints. When i  highlight "Set extruder for selected items" i takes 5 seconds or so for the options to pop up with the extruders.

bubnikv, 2020-03-12:
Would you please make a video of that menu slowness?
We may want to review the callbacks that drive whether the file menu entries are active or grayed out.
I have no idea about the extruder menu slowness.

bubnikv, 2020-03-12:
Actually the file menu slowness may be due to the update of the new Eject SD card / Flash drive. Do you have that file menu slowness in PrusaSlicer 2.2.0-rc2?

Eddyg87, 2020-03-12:
Yes this is all on PrusaSlicer 2.2.0-rc2. I will grab a video later this morning of both issues and update.

Eddyg87, 2020-03-12:
See attached zip file which contains to mp4 videos of selection slowness. Thanks again!

[Slow Menus PrusaSlicer 2.2.0 RC2.zip](https://github.com/prusa3d/PrusaSlicer/files/4323796/Slow.Menus.PrusaSlicer.2.2.0.RC2.zip)

bubnikv, 2020-03-12:
Could you please attach that particular 3MF where the extruder selection was slow?

Do you have many network hard drives connected to your Windows box?

bubnikv, 2020-03-12:
Would you please try the following build?
https://www.dropbox.com/s/dxg35fojkdvw9to/PrusaSlicer-2.2.0-rc3%2B17-win64-g45e007935-202003121544.zip?dl=0

I hope it improves the file menu hanging issue. The difference is, it relies on removable drive enumeration done at a background thread, it no more triggers the removable drive update at the UI thread.

Eddyg87, 2020-03-12:
I have a Synology on the network with 2 "drives" on windows pointing to different locations. The prusa ini are all local on an SSD but that location does automatically sync to the Synology (like Dropbox). I'll try the build and upload the 3MF in a bit.

Eddyg87, 2020-03-12:
Hi Vojtěch,

Tested that build you posted. Fixed the file menu issue and the extruder selection issue. Neither hangs now! Did you still want the 3mf file?

Cheers

Ed

bubnikv, 2020-03-12:
Great that we can close this issue for good.
Thanks for testing.

closed 2020-03-12 by bubnikv