prusa3d/PrusaSlicer #3747: PrusaSlicer crashes during slicing in 'Generic FLEX' setup

a-stark, 2020-03-01:
### Version
Version 2.1.1+win64
(PrusaSlicer-2.1.1+win64-201912101512)

### Operating system type + version
Windows 10 Pro, version 1909, 64bit

### 3D printer brand / version + firmware version (if known)
(I think this is not relevant for this issue)
Brand: Prusa i3 MK3S
Firmware: 3.8.0

### Behavior

The PrusaSlicer crashes (Program just closes) when I try to slice this simple object (attached to this post). It is really a pity, but I hope by pointing out the problem  to improve the slicer performance for the next users.

Here are the steps to reproduce the issue:

* Load the 'Connector_long-improved.STL' into the slicer (drag'n'drop)
* Scale the object from 100% to 70% (in the right panel)
* Select 'Generic FLEX' in the Filament settings (in the right panel)
* Increase the Perimeter contours from 2 (default) to 3 
(here I go to the 'PrinterSettings' --> 'Layers and Perimeters' in the Box 'Vertical Contours'. Hope the translations are correct, since I have the German version.)
* All the other settings are just default settings. Now press 'Slice Now' and the program crashes.

Please note, I have downloaded today the latest version of the PrusaSlicer.

Could you also verify the crash?

#### Project File (.3MF) where problem occurs

The STL file for the slicing (and the .3MF project file):
[Connector_long-improved.zip](https://github.com/prusa3d/PrusaSlicer/files/4272064/Connector_long-improved.zip)

lukasmatena, 2020-03-01:
I reproduced the crash in 2.1.1, but not in 2.2.0-rc. It seems to be already fixed (probably in #2988).

Closing. If you see the crash in any of the current releases, let us know, please.

closed 2020-03-01 by lukasmatena

a-stark, 2020-03-01:
Awesome, thanks for the quick reply! I will wait for the 2.2.0 version for testing.